KasperskyLab/TinyCheck #1: install.sh fails internet connection check

0d0a0c, 2020-11-26:
The line in the internet connection check 

`if nc -zw1 example.com 443; then` 

fails due to the url placeholder example.com

felixaime, 2020-11-30:
Hello again 0d0a0c, 

I don't understand why it fails from your side for "example.com" can you elaborate (some proxies are dropping it?) ? 

Have a good day,
Félix.

posixpoet, 2020-12-01:
Salut a tous
I run in an auth-proxied environment (squid).
Commenting out the check_network in the install file got ride of the network check via nc.

0d0a0c, 2020-12-01:
> Hello again 0d0a0c,
> 
> I don't understand why it fails from your side for "example.com" can you elaborate (some proxies are dropping it?) ?
> 
> Have a good day,
> Félix.

Thanks.

This is very odd! It fails the internet connection on my pi. Also on my laptop using the check_connection function in a terminal.

So I tried example.com in the browser and it fails there too. That was when I changed the URL in the code to google.com which works on my pi and on my laptop. I assumed the issue was that example.com was not a real website, just a place holder in the code.

I have now tried it on another laptop, on the same home network and it works from that laptop. Assuming it is something specific to the affected machines.

closed 2020-12-01 by 0d0a0c

posixpoet, 2020-12-01:
@0d0a0c 
Do you have full understanding and control of the the request path - firewalls, dns, dhcp and proxies - inbetween?
Do you have http_proxy or similar in your env?
Is your browser by any chance pulling a pac file?

In my case - my env is correctly set up. NC, being nc in the install.sh - does not honor env. @felixamie: A simple "curl" maybe smoother than nc. Sorry for not providing a patch at this time. Having other issues with the network interfaces et al.