KasperskyLab/TinyCheck #2: not compatible with ARMv6 raspberry pi models such as pi zero w

0d0a0c, 2020-11-26:
The install.sh script can be tweaked to complete, but nodejs no longer supports ARMv6.

This means lines 258 and 259 in install.sh do not install anything. Options seem to be use an older potentially insecure version of node and npm, cross compile for arm v6, compile directly on the device (I am not even going to try this!) or download somebody else's unofficial and potentially insecure binary.

Will edit this when I've managed to do more with it.

felixaime, 2020-11-30:
Hello 0d0a0c,

Thanks for your involvement and for reporting these first issues, really. 
Sure, I haven't tested it on PiZero, only on standard Raspi 3 and 4. Don't hesitate to tweak it if you think that's faisable. 

Regards,
Félix.

0d0a0c, 2020-11-30:
> Hello 0d0a0c,
> 
> Thanks for your involvement and for reporting these first issues, really.
> Sure, I haven't tested it on PiZero, only on standard Raspi 3 and 4. Don't hesitate to tweak it if you think that's faisable.
> 
> Regards,
> Félix.

Thanks. I will get it running on an ubuntu virtual machine with two real wifi dongles passed from the host.

Once I have that running and get a chance to play with it, I will go back to trying to run it on a pi zero w with an added wifi dongle.

closed 2020-12-01 by 0d0a0c