KasperskyLab/TinyCheck #3: ERR_CONNECTION_REFUSED - cannot access frontend

musafir-py, 2020-11-26:
After successful installation, cannot access the frontend in order to configure tinycheck. I keep getting ERR_CONNECTION_REFUSED error. Tried on two separate networks and routers and few browsers. Same problem. I can ping "tinycheck.local" and resolve IP address of raspberry.

0d0a0c, 2020-11-26:
What are you running it on?

musafir-py, 2020-11-27:
Raspberry Pi 3 B+ with a fresh install of raspberry OS. I tried a direct connection via an ethernet cable with the same result. I can SSH to tinycheck and resolve hostname but no frontend. Also tried changing the frontend config.yml file by switching off kiosk mode and switch on remote access. No progress here as well.

0d0a0c, 2020-11-27:
Ethernet directly to your router should work.

I was able to load the backend in a browser on the pi itself connected to a monitor via hdmi. If you can do this, open chrome and try the address https://127.0.0.1 or https://tinycheck.local

If you can't attach it to a monitor, you could use an ssh tunnel to forward ports 80 and 443 from the pi to the machine you are ssh-ing from (or vice versa, I get confused thinking about ssh tunnels when I haven't used them for a bit), then access the backend on the https://127.0.01 and the front end on http://127.0.0.1 from a browser on the machine you are connecting to the pi from.

The front end should show if you put the local ip address of your pi in the browser of the machine you are accessing the pi from using the ip the pi has on any interface connected to your router, but you can't access the backend remotely without an ssh tunnel or a monitor attached, unless you change the default setting not to allow access to the backend remotely.

I got the front end and backend installed and running on a Raspberry 1 Pi B+, which is armv6 and only 512 mb of ram. When I say running, it is slow, but it works enough to log in to the backend and change settings, and to view the front end. Still some other bits to fix if I am going to get it working on arm v6 but should work on your pi 3.

0d0a0c, 2020-11-27:
> Raspberry Pi 3 B+ with a fresh install of raspberry OS. I tried a direct connection via an ethernet cable with the same result. I can SSH to tinycheck and resolve hostname but no frontend. Also [...]

If you want to connect directly via ethernet cable, from the pi to your laptop or desktop or whatever, you need to modify the dhcp config. In the install.sh script, I think dhcp is set up only for the second wifi interface.

musafir-py, 2020-11-27:
> Ethernet directly to your router should work.
>
> I was able to load the backend in a browser on the pi itself connected to a monitor via hdmi. If you can do this, open chrome and try the address [...]

Thanks. I tried with the fresh full install (last time installed raspberry OS in minimal version). You can access the backend (HTTPS://tinycheck.local) without the problem on another computer connected to the same local network. Frontend (HTTP://tinycheck.local or HTTP://<pi IP address>) gives the same error as last time (even after turning on remote access to frontend on backend panel). 

Accessing frontend on the raspberry browser (HTTP://127.0.0.1) works fine. All functions work without the problem. Only frontend access from another local computer seems to be a problem.

0d0a0c, 2020-11-28:
Perhaps if you kill the backend and frontend services, then run them manually in separate terminals, you will see any error output in the terminal?

felixaime, 2020-11-30:
Hello musafir-py, 

Sorry for the late reply... Regarding the remote access of the front-end you need to enable it from the backend in "**Manage Device > Device config**" and check "**Allow remote access to the frontend**". I'm gonna to create a specific webpage when the user is not authorized to access remotely to the frontend to present him the steps to do. 

Don't hesitate if you have more issues.
Have a good day,
Félix.

musafir-py, 2020-12-18:
Problem solved. All working. Thanks.

closed 2020-12-18 by musafir-py