KasperskyLab/TinyCheck #6: issues with generating network on raspberry pi 3+

mishoops, 2020-12-01:
I'm having the following issue:

```
Unfortunatelly, we got some issues.

Please verify that you've to Wifi interaces on your device and restart it by clicing on the bottom below
```

I've checked syslog and got following information when I click to use existing network 

```
Dec  1 14:30:19 tinycheck dhcpcd[484]: wlan1: carrier acquired
Dec  1 14:30:19 tinycheck kernel: [  304.598639] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Dec  1 14:30:19 tinycheck dhcpcd[484]: wlan1: IAID eb:d1:47:a4
Dec  1 14:30:19 tinycheck dhcpcd[484]: wlan1: adding address fe80::6e18:8148:cde4:2469
Dec  1 14:30:19 tinycheck avahi-daemon[401]: Joining mDNS multicast group on interface wlan1.IPv6 with address fe80::6e18:8148:cde4:2469.
Dec  1 14:30:19 tinycheck avahi-daemon[401]: New relevant interface wlan1.IPv6 for mDNS.
Dec  1 14:30:19 tinycheck dhcpcd[484]: wlan1: probing address 192.168.100.1/24
Dec  1 14:30:19 tinycheck avahi-daemon[401]: Registering new address record for fe80::6e18:8148:cde4:2469 on wlan1.*.
Dec  1 14:30:20 tinycheck dhcpcd[484]: wlan1: soliciting an IPv6 router
Dec  1 14:30:20 tinycheck python3[396]: 127.0.0.1 - - [01/Dec/2020 14:30:20] "GET /api/network/ap/start HTTP/1.1" 200 -
Dec  1 14:30:25 tinycheck dhcpcd[484]: wlan1: using static address 192.168.100.1/24
Dec  1 14:30:25 tinycheck avahi-daemon[401]: Joining mDNS multicast group on interface wlan1.IPv4 with address 192.168.100.1.
Dec  1 14:30:25 tinycheck avahi-daemon[401]: New relevant interface wlan1.IPv4 for mDNS.
Dec  1 14:30:25 tinycheck avahi-daemon[401]: Registering new address record for 192.168.100.1 on wlan1.IPv4.
Dec  1 14:30:25 tinycheck dhcpcd[484]: wlan1: adding route to 192.168.100.0/24
Dec  1 14:30:32 tinycheck dhcpcd[484]: wlan1: no IPv6 Routers available
```

rfkill shows 

```
root@tinycheck:/var/log# rfkill list all
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no
2: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
root@tinycheck:/var/log#
```

and ifconfig

```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.88.68  netmask 255.255.255.0  broadcast 192.168.88.255
        inet6 fe80::5201:2c77:4fe2:f1f9  prefixlen 64  scopeid 0x20<link>
        ether d0:37:45:ff:8f:df  txqueuelen 1000  (Ethernet)
        RX packets 29889  bytes 3631819 (3.4 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1467  bytes 800955 (782.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.100.1  netmask 255.255.255.0  broadcast 192.168.100.255
        inet6 fe80::6e18:8148:cde4:2469  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:d1:47:a4  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 37  bytes 5172 (5.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

**Setup**:

Raspberry Pi 3+
TP-Link TL-WN725N
```
root@tinycheck:/var/log# uname -a
Linux tinycheck 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
root@tinycheck:/var/log#
```

felixaime, 2020-12-01:
Hello mishoops,

Thanks for the feedback. 

Its strange. Can you show me your "/tmp/hostapd.log" after trying to generate a network? This message is just a "wrapper" if there is some errors with hostapd - eg. in network.py:

`if path.isfile("/tmp/hostapd.log"):
                with open("/tmp/hostapd.log", "r") as f:
                    log = f.read()
                    err = ["Could not configure driver mode",
                           "Could not connect to kernel driver",
                           "driver initialization failed"]
                    if not any(e in log for e in err):
                        if "AP-ENABLED" in log:
                            return True
                    else:
                        return False
`

I think that I have the same wifi adapter @home, I'm gonna try these next days.

mishoops, 2020-12-01:
Hi @felixaime 
Thank you for quick reply

```
root@tinycheck:/var/log# cat /tmp/hostapd.log
Configuration file: /tmp/hostapd.conf
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan1: Could not connect to kernel driver
Using interface wlan1 with hwaddr b8:27:eb:d1:47:a4 and ssid "network-0aa0"
wlan1: interface state COUNTRY_UPDATE->ENABLED
wlan1: AP-ENABLED
root@tinycheck:/var/log#
```

mishoops, 2020-12-01:
there is also a config file created to it

```
root@tinycheck:/var/log# cat /tmp/hostapd.conf
country_code=GB
interface=wlan1
ssid=network-0aa0
hw_mode=g
channel=7
auth_algs=1
wpa=2
wpa_passphrase=c62197d9
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
disassoc_low_ack=0root@tinycheck:/var/log#
```

wireless country code is set to GB, is this correct? it shouldnt be as im not in GB 
 
```
disassoc_low_ack=0root@tinycheck:/var/log# iw reg get
global
country GB: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

root@tinycheck:/var/log#
```

mishoops, 2020-12-01:
OK, so i've manually connected to wifi as per settings from /tmp/hostapd.conf and disabled LTE and GSM network in general and i do get connections. 
Where does the system store PCAPs ?

felixaime, 2020-12-01:
In `/tmp/[A-F0-9]{8}/`, for each capture you have a new folder created with all the stuff (pcap, logs etc.). 

Regarding the "GB", you can change it under `/usr/share/tinycheck/server/frontend/app/assets/hostapd.conf`. As of today, I don't allow the user to choose the country but I can dev it for a next release.

mishoops, 2020-12-01:
so no folder as such was created when i run it. It created both /tmp/hostapd.{log,conf} but not a folder. 

I have the following flow

Go to app page -> click Let's Start -> click Yes, use it -> We are generating network for you (this step creates above files) -> and than "Unfortunatelly we got some issue"
If instead of reboot i refresh the page, i can go through the same process and new network will be created. 

Could it be some issue with frontend not generating QR code? (wild guess) 

```
pi@tinycheck:~ $ ls -al /tmp/
total 56
drwxrwxrwt 11 root root 4096 Dec  1 18:50 .
drwxr-xr-x 22 root root 4096 Dec  1 13:34 ..
drwx------  2 pi   pi   4096 Dec  1 18:50 dhcpcd-pi
drwxrwxrwt  2 root root 4096 Dec  1 18:49 .font-unix
-rw-r--r--  1 root root  186 Dec  1 18:50 hostapd.conf
-rw-r--r--  1 root root  524 Dec  1 18:51 hostapd.log
drwxrwxrwt  2 root root 4096 Dec  1 18:49 .ICE-unix
drwx------  2 pi   pi   4096 Dec  1 18:49 ssh-07XxFkBXAU9z
drwx------  2 pi   pi   4096 Dec  1 18:49 ssh-gC0rKQTEPD1y
drwx------  3 root root 4096 Dec  1 18:49 systemd-private-e05c46cdffa54b8995a94f741abefbff-systemd-timesyncd.service-99BWXE
drwxrwxrwt  2 root root 4096 Dec  1 18:49 .Test-unix
-r--r--r--  1 root root   11 Dec  1 18:49 .X0-lock
drwxrwxrwt  2 root root 4096 Dec  1 18:49 .X11-unix
drwxrwxrwt  2 root root 4096 Dec  1 18:49 .XIM-unix
```

posixpoet, 2020-12-01:
@mishoops 
I suspect some python modules missing.
I had to do the following after a raspian-lite install:

* Installed raspbian-lite on a Pi4; Edimax USB WIFI Plug & onboard WIFI

** Network check fail
Running in a (auth) proxied network?
nc in install.sh - comment out the check_network. 

** Network card names
raspi-config: Disable predictable names

** Manual install of:
apt-get install python3-pip
 python3-flask
 python3-sqlalchemy
 python3-flask-httpauth
 python3-yaml
 python3-jwt
 python3-netifaces
 python3-qrcode
pip3 install wifi psutil pyudev netaddr ipwhois pydig

mishoops, 2020-12-02:
@posixpoet thank you for tips, i've installed packages but still no luck. Can you please tell me which edimax USB dongle did you use?
Edimax EW-7811Un or something else?

posixpoet, 2020-12-02:
@mishoops Yes, lsusb reports it as EW-7811Un.
Both WIFIs are detected and setup (ip a), right?

mil1234n6, 2020-12-27:
Did anyone face the issue that the page says; creating Ap.. but nothing happens after? When i list my wifi’s i see that the ap is creates but i cant proceed to the next step it seems.

would love to try out this project

felixaime, 2020-12-29:
Hello, its the same issue as the #28 ?

mil1234n6, 2020-12-29:
Yes @felixaime

mishoops, 2021-01-05:
> Hello, its the same issue as the #28 ?

Thanks, this solved the issue.

closed 2021-01-05 by mishoops