KasperskyLab/TinyCheck #7: Uninstall instructions

skanga, 2020-12-03:
I tried it out on Ubuntu 20.04.1 LTS which is the only system which had 2 wifi interfaces. The installation gave a lot of errors and then rebooted. Now I cannot connect to the web interface - although ssh still works. 

I want to uninstall TinyCheck - how can I do that?

felixaime, 2020-12-04:
Hello Skanga,

The shortest way is to do:

`# rm -rf /usr/share/tinycheck/*`
`# rm /lib/systemd/system/tinycheck*`
`$ rm ~/Desktop/tinycheck.desktop`

and edit your `/etc/hostname` & `/etc/hosts` with the name that you want to give back to your system. 

After, depends if you need to delete libraries and dependencies.

skanga, 2020-12-05:
Thanks @felixaime 
That mostly worked ...
Now whenever I run an apt update it throws an error

E: The repository 'http://download.opensuse.org/repositories/security:/zeek/Debian_20.04  Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

skanga, 2020-12-05:
I deleted /etc/apt/sources.list.d/security:zeek.list and it went away ...

linbbo70, 2020-12-22:
Hi, Tinycheck cannot find Chromium. I tried renaming the browser and it's commands, but without success. If there is a way to open it in the browser that I am missing, I will reinstall it.

Thank you,

felixaime, 2020-12-29:
Hi @linbbo70,

If your issue is not solved regarding your chromium stuff, please open another issue. Normally, it can work on a standard Raspi host which have chromium installed by default. 

Félix.

closed 2020-12-29 by felixaime

linbbo70, 2020-12-29:
Hi, I'm using Kubuntu 20.10. I installed chromium and then TinyCheck which rebuted without a problem. Clicking on its icon I get the message:
Unknown error code 100
Could not find the program 'chromium-browser'
Please send a full bug report at https://bugs.kde.org.
I have tried renaming chromium-chromium to chromium-browser, but no changes seem to work.
Thanks, Ray