KasperskyLab/TinyCheck #10: Why two WiFi dongles required?

dazzag24, 2020-12-04:
Hi,

If you have RaspberryPi connected via ethernet cable, could you get by with only a single WiFi dongle?

Thanks

felixaime, 2020-12-04:
Hello dazzag24, 

Yes, but it is not implemented yet (simply that I haven't access to an ethernet cable for that... and still haven't). In that case you need to change in config.yaml the network out by eth0. I think that it will work correctly. 

Félix.

dazzag24, 2020-12-04:
Hi

Thanks for the reply. I'll see if I get a chance to test as I think many
people will have at least one WiFi dongle and access to a wired connection.

Darren

felixaime, 2020-12-04:
Yeah sorry. There is a miscomprehension here I think. Its only ONE wifi dongle, two wifi interfaces (the second interface is the one of your raspberry.)

tg12, 2020-12-08:
Yeah, Any interface would be a brilliant addition, You could even use this in larger envs if you have Port Mirroring or even a networking hub on a home network.

Mycah, 2020-12-21:
I'm trying this.  So far I've changed line 312 of install.sh:
`if [[ "${#rfaces[@]}" == 1 ]]; then`
So that it will install with only one wlan interface.
Then updated line 56 of config.yml:
`out: eth0`
Once that's done, I run the installer.  Everything goes fine, and I see no errors.
After reboot, tinycheck.local says I'm already connected to a network, and asks if I want to use it, I say yes.
Then it pops this up.
[image]
After a reboot it does the same thing.
I'm not sure where the logging is to identify how to circumvent this issue.
I do see in my /usr/share/tinycheck/config.yml it does have this network config:
`network:
  in: wlan0
  internet_check: http://example.com
  out: eth0
  ssids:
  - wireless
  - skynet
  - network
  - wifi
  tokenized_ssids: true
`
I want to try this tool so I'm happy to continue troubleshooting if you have further suggestions.

felixaime, 2020-12-22:
Hello !

Thanks you very much for you feedback. Regarding this error message (that I need to change because nobody understand what's going on...) its related to hostapd when it throws an error. Is is possible to get the content of the file`/tmp/hostapd.log` (which contains the full error stack) and the result of the command `iw wlan0` to see the modes accepted by your Wi-Fi interface ? Moreover, you run it on what ? A raspberry pi ?

felixaime, 2021-02-19:
Hello, 

I think that with the docs and the use of Ethernet available, everything is solved.
Case closed for me, do not hesitate to re-open it if needed.

Have a good week,
Félix.

commit 14fec393 (2022-06-04): Fixes #10
closed 2021-02-19 by felixaime