KasperskyLab/TinyCheck #11: Running headless?

Mycah, 2020-12-05:
Most of my RPI's haven't had a monitor connected at any time, and I'd love to be able to use this as a auditing tool on one of them.

Is there a way to install this headless without a running x11?  I tried a couple of times, and it appears it won't run without a desktop loaded.

Thanks!

posixpoet, 2020-12-06:
Odd - I installed it exactly that way, using raspian-lite.
Now, I trying to get kiosk mode working (with a 5' TFT).

felixaime, 2020-12-10:
Hello Mycah,

Sorry for the late reply, yeah of course you can. Install it on your raspberry pi and after you can connect to it via a browser by going to https://tinycheck.local (backend) http://tinycheck.local (frontend). Regarding the desktop, I need to verify the services but I think that only the kioskmode needs the desktop, the other services can run without desktop setted.

closed 2020-12-29 by felixaime