KasperskyLab/TinyCheck #12: [Documentation] The WIFI dongle used by hostapd must be in monitor mode

jbrinksmeier, 2020-12-05:
Hi felix, first let me say that I'm entirely grateful for the work you put in that project. Much appreciated :+1: 

While setting this up on a PI 4+ with raspbian OS I stumbled across something that is worth to be mentioned in the readme imo.
The WIFI (dongle) used to setup the client's networks via hostapd needs to be running in monitor mode. While this might be obvious for the folks that hack with wifis regularly, it might be a hassle for others to finally find out. Especially when (as in my case) the default drivers do not support it. Would you accept a PR for this?

Best regards

felixaime, 2020-12-06:
I think that's more Master mode, no?

I'm gonna try some stuff to have a better documentation regarding that (and list of wifi adpaters that are working as it seems a big issue). Btw its possible to look at your raspberry pi wifi interface, maybe it can support master mode so you have just to switch the two interfaces in the configuration...)

Sorry for the inconvenience...

jbrinksmeier, 2020-12-06:
well, to my knowledge in nl80211it's called monitor mode (referring to https://help.ubuntu.com/community/WifiDocs/MasterMode), but not worth to be nitty-gritty here I guess :)

My finding on this was that both the dongle I used and the on-board wifi of the PI are running in "managed" mode, which is unsuitable for the usage with hostapd, which requires modes "monitor" or "AP". Both modes where unavailable with the pre-installed drivers in raspberryOS and required me to build and install some other drivers. I ended up using https://github.com/AIRCRACK-NG/RTL8188EUS for my dongle, switched the device to "monitor" mode and tinycheck started working with minor fixes which I plan to address in a separate issue if applicable . 

> Sorry for the inconvenience...

No problem at all. I think for a first release of a tool that relies on such a certain hardware setup it works just great. I analyzed the first phones today and it's really fun and helpful :)

felixaime, 2021-01-07:
Back on that thread after some holiday, it is AP mode for the interface as you don't need to sniff any packet. 

When you run the AP, you can check that the interface is under AP mode by using the command "iw dev". 
No need monitor capabilities to run it.

Félix.

closed 2021-01-07 by felixaime