KasperskyLab/TinyCheck #14: [Documentation] What's the maintainance plan with iocs.json

jbrinksmeier, 2020-12-06:
While searching for possible extensions to the IOCs used by tinycheck, I found myself missing some information about how this iocs.json is to be maintained by this project. You mentioned some sources in the docs, for example https://github.com/Te-k/stalkerware-indicators, but it looks like not all of the available iocs from there made it into the iocs.json. I am creating a watcher for them right now and it is painfree with the architecture you came up with.

In general it would be good to know how plans are to maintain the iocs.json in this repo. Are you watching sources like Te-k/stalkerware-indicators proactively and update the iocs.json? Is the plan to maintain this repository as a comprehensive list of IOCs with input/PRs from the community as 

> If you have seen something very suspicious and/or needs to be investigated/integrated in one of these two lists, don't hesitate to ping us. You can also do you own watcher. Remember, sharing is caring.

suggests?

Thanks for a feedback on this matter :)

posixpoet, 2020-12-06:
Yes, maintenance or better: "transparency" of the utilized IOCs would be great. 
I imagine a Info page and part of the report (if not already).

@felixaime : It's been said before - great work. My first runs with it are promising.

felixaime, 2020-12-08:
Hello jbrinksmeier,

> While searching for possible extensions to the IOCs used by tinycheck, I found myself missing some information about how this iocs.json is to be maintained by this project. You mentioned some [...]

Yes you're right. Strange for the Tek repo, I though to have integrated all of them. We are thinking to do a special export on another github repo centralizing only stalkerware IOCs and maintained by guys working on this kind of threats.  

> In general it would be good to know how plans are to maintain the iocs.json in this repo. Are you watching sources like Te-k/stalkerware-indicators proactively and update the iocs.json? Is the plan [...]

Really, IOCs management is still a pain in the ass (for example, I'm not mentioning the source, validity period associated to them etc.). I wanted something very easy and small to begin. As of today, trust me that there is no "strategy" to maintain it except inputs/PRs from the community and myself. I came with this bucket of few IOCs to launch the projet. 

I'm thinking to create a Wiki page on them (we definitely need a wiki-like stuff to share ideas and improvements). Anyway, here is a list of what's integrated:

**To hunt the "unknown" threats:**
- Many known dynamic/free DNS from AFRAID and other DynamicDNS providers;
- Different fancy top level domains (such .xyz, .club etc.);
- Network ranges associated to some hosters known to be APTs nests;
- Name servers associated to bulletproof domain services;
- Snort rules (much like experiments here for now.)

**To hunt the "unknown" threats:**
- Tek and Cian stalkerware network IOCs (I know that they're not exhaustive)
- Costin geo-trackers list from his repo.
- Emilien's stratum rules to find some miners.

> 
> Thanks for a feedback on this matter :)

You're welcome!

jbrinksmeier, 2020-12-09:
great summary, thank you. that's been exactly what I wanted to know. 
Meanwhile, I created a PR for Tek's IOC repository that creates a file for each release including the known domains and certificates to be included by adding `https://github.com/Te-k/stalkerware-indicators/releases/download/indicators-for-tinycheck.json`to the ioc watchers in `config.yml`. Maybe this is of some use to others while this project evolves. 
If you find the time, I'd be grateful for you quickly checking if I got the format of the ioc file right here: https://github.com/Te-k/stalkerware-indicators/releases/download/indicators-for-tinycheck.json. I added samples in the description. 
I'm especially unsure about the `tpl` property as I found no usages of them and simply hardcoded them to `white` as the others in the original ioc.json.

felixaime, 2020-12-10:
Your URL doesn't work yet :/ After that, yeah I can check and integrate the URL to the default watchers ;)

Félix.

jbrinksmeier, 2020-12-10:
ups, my bad. The url is obviously not working as the PR is not been accepted yet. Indeed I wanted to link to the PR: https://github.com/Te-k/stalkerware-indicators/pull/5

Te-k, 2020-12-12:
I have updated the Github action to build the JSON file automatically, thanks @jbrinksmeier ! So I think anyone can use that `https://github.com/Te-k/stalkerware-indicators/raw/master/indicators-for-tinycheck.json` as a source for tiny check.

If I am right the certificates in the TinyCheck list are TLS certificates, not android app certificates? If yes, it does not make sense to have the android app certificates in this list, but I can do some work to have a list of IP addresses if it helps Tiny Check detection.

Regarding IOCs, I have tried to integrate in this repository all the indicators I have seen from different public sources, so it should be quite complete. I am totally open to add some more if you see some missing indicators (and I am going to do some more reversing soon to get some more domains)

felixaime, 2020-12-13:
Cool! Thanks Te-k! I'm gonna to test it and integrate it in the watchers this week if I have time!

Have a good sunday,
Félix.

felixaime, 2020-12-16:
Hello,

I've added the feed this evening to the default watchers IOCs. Thanks Te-k for your contribution!
I keep this issue open for anyone who have questions about IOCs.

Have a great evening,
Félix.

posixpoet, 2020-12-21:
Oh mighty wizards
I may be asking much, still: Would the IOCs be able to discover NSO Group’s Pegasus spyware?
https://citizenlab.ca/2020/12/the-great-ipwn-journalists-hacked-with-suspected-nso-group-imessage-zero-click-exploit/
Stay safe,
me

felixaime, 2020-12-21:
Hello posixpoet,

I think yes (Medium alerts at least) because their domains used some nameservers which are already referenced in the iocs.json. I hesitate to add the certificates issuers, but I think that it will be more FP than Let's Encrypt. 

I'm in holiday so I haven't the time to investigate more deeply on their case. 
Cheers,
Félix.