KasperskyLab/TinyCheck #15: Saving a report to USB

posixpoet, 2020-12-07:
As anyone been able to do this through the frontend?
I select "save", then plug in a USB stick - but nothing else happens beyond that.
(I am expecting an auto-mount of the USB as a next step)

I assume the /tmp/CAPTURESESSION/*.json and the logs are part of the report.
Can I trigger the save manually from "...tinycheck/server/frontend/app/classes/save.py"?

TIA

felixaime, 2020-12-08:
Hello,

If I understand well, nothing is saved on your usb stick? Does the USB stick is mounted automatically by the system? 
('cause I'm just looking of mounted USBs drives in the code, I don't mount anything myself...)

And regarding for the report, yeah, it saves as a ZIP the full content of this directory on the mounted USB stick. As of today, I don't have a pretty PDF/HTML report but just a json...

Félix.

posixpoet, 2020-12-08:
Found it!
Package _usbmount_ was missing from my raspbian-lite install.
I also had to change "PrivateMounts=yes" to "PrivateMounts=no" in /lib/systemd/system/systemd-udevd.service.

closed 2020-12-08 by posixpoet