KasperskyLab/TinyCheck #19: Decrypting TLS encrypted traffic

ranlo, 2020-12-14:
Pardon the ignorance - is there any way in which TinyWatch can decrypt https requests? 
Obviously, this would require installing a certificate on the device, so it require the OK of the device owner. But for anyone analysing the traffic and trying to hunt for intrusions, the additional data (endpoints getting called; strings getting sent/received) would be very beneficial. All these are currently 'invisible' if they are sent over https

jbrinksmeier, 2020-12-21:
For what it's worth, I don't like the idea of tinycheck being that invasive on target devices. From my point of view, tinycheck is not an investigators tool but one that gives users a first check whether their device needs a more thorough investigation and a starting point for analysts.

posixpoet, 2020-12-21:
@jbrinksmeier , highly agree. There are other methods to investigate this TLS-deep.
There is a high risk of over-engineering this while missing basics.
For me, basics (top of my head) are:

- Transparent IOCs management
- Robust kiosk/lender mode (power loss, tamper and data loss proof,...)

It would be devestating, this device to be lost/stolen with information/assessment fragments left on it.

ranlo, 2020-12-21:
Fair enough, valid concerns. Thanks for the feedback!

closed 2020-12-21 by ranlo