KasperskyLab/TinyCheck #20: [Feature] TinyCheck should have localization

jbrinksmeier, 2020-12-18:
While english is a totally fine thing to start with, it would be good if the interfaces of tinycheck had the possibility to be translated. 

I'll leave this as idea for now. Maybe I find time this month to work on that myself. Also I would be happy if people share possible solutions for that here. Right now we have stuff to translate in both frontends and the analysis engine.

felixaime, 2020-12-18:
Hello !

Yeah for sure, we were thinking on that today during a meeting. I have to see how its possible easly. Today we need to translate the alerts (which are hardcoded but I'll improve that because hardcoding stuff is shit) and the GUI. I'll came back with solution in January regarding that. Anyway, if you want to help, even just by translating some stuff, don't hesitate !

Félix.

jbrinksmeier, 2020-12-21:
I'll give it a try and either implement https://www.npmjs.com/package/vue-translate-plugin or https://github.com/kazupon/vue-i18n for the frontend. Right now, our use case does not need translations for the alerts (in fact, it could be counter-productive for us not having them in english). I'll add german translations. Let's see if this is useful enough for the project to be merged. If not, we would at least have the german translation.

felixaime, 2021-02-08:
Hello @jbrinksmeier !

After few weeks, I'm gonna to publish a new version with the support of multi-languages thanks to the help of @mxgnr. As of today, only two languages are implemented (French and English). If you want to participate in the translation of the project, you have two translate two different files:

- /app/frontend/src/locales/en.json - Frontend interface
- /analysis/locales/en.json - Alerts

Have a good week!
Félix.

felixaime, 2021-02-12:
I close the thread :)
Don't hesitate if you want to add more languages ;)

Félix.

closed 2021-02-12 by felixaime