KasperskyLab/TinyCheck #25: Frontend / admin: One SSID breaks config.yaml formatting for "network:"

posixpoet, 2020-12-22:
When removing all SSIDs unter admin page "MANAGE DEVICE/Network config" and leaving only one, the config.yaml modified.
The modification is subesquently parsed errornously.

Previous setting under "network:/ssids:" (config.yaml):
 ...
 network:
  in: wlan1
  internet_check: http://example.com
  out: wlan0
  ssids:
  - skynet
  - wireless
  tokenized_ssids: true
 ...

After removal of all and addtion of SSID "TEST":
 ...
 network:
  in: wlan1
  internet_check: http://example.com
  out: wlan0
  ssids: TEST
  tokenized_ssids: true
 ...

"TEST" is subsequently then rendered as SSIDs "T", "E", "S", "T".
Expected behaviour is restored when manualyl editing config.yaml to
 ...
  ssids: 
  - TEST
 ...

felixaime, 2020-12-27:
Thanks for the reporting, I've done a commit which is correcting this issue and others related to the configuration API.
https://github.com/KasperskyLab/TinyCheck/commit/86193d912b93a2a598393f3235f5264d5bc753ab

Thanks for your feedback!
Félix.

posixpoet, 2020-12-27:
Well done!
(Sorry, didn't offer the patch myself. Will do better.)

closed 2020-12-27 by posixpoet