KasperskyLab/TinyCheck #26: Use of TinyCheck without internet access

leandreSL, 2020-12-22:
I wonder if it's possible to use TinyCheck without a "real" internet access. The purpose is, if the smartphone is infected with a spyware, to not send more information to any C2 server.
I can also use a "fake internet" with fake DNS/HTTP responses

felixaime, 2020-12-29:
Yeah it can be technically possible by using something like Fakenet (or just some netfilter rules and a fake DNS server). Maybe it can be even implemented directly on the TinyCheck device. It can be good to dig on that.

felixaime, 2021-03-01:
Hello,

Now TinyCheck can run without Internet by forcing this feature if TC is not connected to any network. The only issue is that the analysis will be only based on domain names. Good for stalkerware, bad for the other threats :)

Have a nice day,
Félix.

closed 2021-03-01 by felixaime