KasperskyLab/TinyCheck #28: AP gets created, but tinycheck doesnt proceed

mil1234n6, 2020-12-28:
Hey,

Ive installed TinyCheck on my Rasberry Pi zero W. 
And after some struggles to get it all working it now stops working at the point where it should show the QR code with created AP name and password.  I can connect to the AP, also have a connection. But tinycheck doesnt go on..

I watched the debugger in firefox and it shows:

Uncaught (in promise) Error: timeout of 30000ms exceeded
    exports createError.js:16
    ontimeout xhr.js:103

So thats probably this part?:

methods: {
        generate_ap: function() {
            clearInterval(this.interval)
            this.ssid_name = false
            axios.get(`/api/network/ap/start`, { timeout: 30000 })
                .then(response => (this.show_ap(response.data)))

Any clues what i could do? Is it maybe the QR code what doesnt get generated or?
Thanks in advance.

mil1234n6, 2020-12-28:
It does seem to work when i changed the timeout from: 30000 to 300000.

felixaime, 2020-12-28:
Hello! 

On what device / os are you running TinyCheck? Strange to see that it takes so much time to proceed. Normally it takes like 3/4 seconds to generate the AP + QRCode.

Félix.

mil1234n6, 2020-12-28:
Hey Félix,

On a rasberry pi zero w running latest rasberry pi os.
Its not taking 'anoyingly' long. Ive counted and its like: 40-50~ seconds
The other steps are working perfectly fine.

Only problem now after i rebuild it I cant access the backend anymore...

mil1234n6, 2020-12-28:
Ah nevermind the new problem :) Ive ran update.sh aswell. That does remove the rsa keys but doesnt generate new ones. So after running `openssl req -x509 -subj '/CN=tinycheck.local/O=TinyCheck Backend' -newkey rsa:4096 -nodes -keyout /usr/share/tinycheck/server/backend/key.pem -out /usr/share/tinycheck/server/backend/cert.pem -days 3650`
it worked again!

felixaime, 2020-12-28:
Roger that, I'm updating the update.sh script with that and other stuff. 
Thanks for reporting that too.

felixaime, 2020-12-29:
> Hey Félix,
> 
> On a rasberry pi zero w running latest rasberry pi os.
> Its not taking 'anoyingly' long. Ive counted and its like: 40-50~ seconds
> The other steps are working perfectly fine.
> 
> Only problem now after i rebuild it I cant access the backend anymore...
> 
> Thanks for your fast response!

I'm gonna to update the `/api/network/ap/start` timeout to make it working. But it can be interesting to see from where there is the bottle neck. Can you test juste to start an hostapd instance from the command line an see how many times it takes ?

Cheers,
Félix.

mil1234n6, 2020-12-29:
Hey Felix,

The AP is created very quickly, can see that when i open up available wifi's. Not more than a few seconds.
I was wondering, before it did work and i connect, i still could surf on my phone using that wifi (I believe). If i connect now using for example the QR code. im connected but dont have internet. I was wondering whats the right behaviour now

anna7x, 2020-12-29:
Hey,

I have installed TinyCheck on my Ubuntu 18 vm , did not crate the AP ,struggles to get work but no luck.

i used comfast usb wireless  cf-wu710n v2

the debugger in firefox and it shows 

POSThttp://tinycheck.local/api/network/wifi/setup
[HTTP/1.0 500 INTERNAL SERVER ERROR 7ms]

Error: Request failed with status code 500
    exports createError.js:16
    exports settle.js:17
    onreadystatechange xhr.js:69
    exports xhr.js:44
    exports xhr.js:13
    exports dispatchRequest.js:52
    promise callback*0a06/u.prototype.request Axios.js:61
    e Axios.js:86
    exports bind.js:9
    wifi_setup wifi-select.vue:104
    click wifi-select.vue:1

mil1234n6, 2020-12-30:
@anna7x When u installed it, did u see that the installation also build the front/backend?, or when you check your front/back end directories. Do you see the directory Dist?

mil1234n6, 2020-12-30:
@felixaime The fix worked. ive ran the update.sh again and i dont get the timeout again.

xb8, 2021-01-08:
i have the same problem, i can generate an AP (in about 6 seconds), connect to it, then tinycheck gets stuck and the phone i connect to it doesn't have internet connection from that AP

felixaime, 2021-01-14:
Is it an iPhone ? I have this issue sometime with the iPhones by using the QRCode. I stop the capture, recreate one and that's OK. Maybe the /tmp/hostapd.log will told you the truth of what's going on.

felixaime, 2021-02-19:
No answer for days, issue closed.

closed 2021-02-19 by felixaime