KasperskyLab/TinyCheck #31: Unable to browser tinycheck

vespertillo, 2021-01-02:
Hi, i install utility on Kali everythings fine but After reboot when try to open  browser the address  http://tinycheck.local dont work, get isp error site dont exist. Its strange that dont open host tinycheck any help?

felixaime, 2021-01-04:
Ok, can you test please http://127.0.0.1 ?

vespertillo, 2021-01-04:
ok i forget pc and try on raspberry,but get error when launch app need 2 device but already connecting.
Wifi of device is connected to my wifi,2 dongle not but when try to congigure get this error
[image]
after reboot same error any help? im newbie how to find and analyze error file?

vespertillo, 2021-01-06:
i try to run update.sh  but same error.

felixaime, 2021-01-07:
Hello vespertillo,

Sorry for the late answer (Holidays...). Ok, if you have that error, can you send:
   - The content of `/tmp/hostapd.log` (after an unsuccessful launch) 
   - The result of the command `ifconfig` (to see your wifi interfaces)
   - The result of the command `iw list | grep -A 8 "Supported interface modes"`

If you're not very confortable to publish it here, don't hesitate to send them at tinycheck--@--kaspersky--.--com or contact me directly via twitter (@felixaime).

xb8, 2021-01-07:
Raspberry  pi 4 model b, fresh kali for pi install, tinycheck doesn't work. 2 problems: the desktop link doesn't find chromium, but regardless of that nothing shows up at 127.0.0.1, both on http:// and https:// so neither the frontend and the backend

vespertillo, 2021-01-08:
my hostapd log im not GB
country_code=GB
interface=wlan1
ssid=network-a21f
hw_mode=g
channel=7
auth_algs=1
wpa=2
wpa_passphrase=2901c105
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
disassoc_low_ack=0

ifconfig
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.104  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::131:6b89:71ea:5399  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:9e:b6:de  txqueuelen 1000  (Ethernet)
        RX packets 2384  bytes 139116 (3.4 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1692  bytes 256410 (782.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4099<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether 8e:4e:06:19:ad:dc  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

iw list | grep -A 8 g

-pi@tinycheck:~ $ iw list | grep -A 8 g
	max scan IEs length: 2285 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports RSN-IBSS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
--
		 * managed
		 * monitor
	Band 1:
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
--
		 * set_mesh_config
		 * set_bss
		 * authenticate
		 * associate
		 * deauthenticate
		 * disassociate
		 * join_ibss
		 * join_mesh
[... 128 lines cut ...]
			* 48.0 Mbps
--
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	software interface modes (can always be added):
	valid interface combinations:
		 * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
		   total <= 3, #channels <= 2
		 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
		   total <= 4, #channels <= 1
	Device supports scan flush.
	Device supports randomizing MAC-addr in sched scans.
	Supported extended features:
		* [ 4WAY_HANDSHAKE_STA_PSK ]: 4-way handshake with PSK in station mode
		* [ 4WAY_HANDSHAKE_STA_1X ]: 4-way handshake with 802.1X in station mode
		* [ DFS_OFFLOAD ]: DFS offload

vespertillo, 2021-01-09:
i use raspberry p3+, fresh install rom from raspberry site

felixaime, 2021-02-19:
Hello,

Have you tested to re-install it with the latest versions ?

Félix

vespertillo, 2021-05-17:
No, i try this week

fixpc82, 2022-07-04:
Is possible to install on a pc with Ubuntu 18?

HonourableTyr, 2022-07-17:
> Is possible to install on a pc with Ubuntu 18?

Not really. It isn't designed to be deployed on x86 systems. I did try to deploy it on PiOS x86 at first. It may be possible but you'd have a hell of a lot of work. Potentially having to change a lot of the dependencies.

vespertillo, 2022-09-21:
Hi i reinstall using Raspberry os 64 kernel 5.15 Debian 11 bullseye,during install have error when script install dependecies m2crypto,install pass but stay many hours on downloading iocs any help?After install os dont update It directly install tinycheck need update os after install?