KasperskyLab/TinyCheck #35: Frontend in loop, does not proceed

githlp, 2021-01-23:
How to reset all initial settings ?

After changing the internet access from eth0 to wlan1 the frontend website (http://tinycheck.local) shows the logo and a loading circle gif but nothing else happens..

I already stopped the FE/BE/Watcher service, started them again and checked that status is ok..

I would like to restart over with the screen that suggests a new wifi network with QR and network name/pass.

what files do i have to edit ? thank you for your support.

githlp, 2021-01-23:
With new installation of RaspberryPi OS and TinyCheck the program now creates an AP but other devices cannot connect/obtain ip address ... seems to be a problem for the edimax wifi dongle to act as AP, will try others.

felixaime, 2021-01-25:
Hello,

Can be interesting to see :with route what's your default route interface and if its equal to the network > out value in the config. The same,

felixaime, 2021-01-25:
I'm back here,

Related to "After changing the internet access from eth0 to wlan1 the frontend website (http://tinycheck.local) shows the logo and a loading circle gif but nothing else happens.."

I've found an error in `splash-screen.vue` which can lead to an infinite loop if you're using wifi to wifi and your raspberry pi is not connected to Internet. I'm correcting that this afternoon.

If you want to change the interfaces, don't do it from the configuration but from the administration panel at https://localhost (if you're using it locally) or https://tinycheck.local. Indeed, changes will be done not only in config.yaml but also `dnsmasq.conf` and `dhcpcd.conf`.

githlp, 2021-01-25:
Thanks a lot for quick reply!

Will try with different usb wifi device asa i got it and then report back.

githlp, 2021-01-28:
Hello,
well even after hours of trying it does not work out. The configuration of wifi seems to be completely destroyed.

I tried with 4 different USB wifi adapters, and none of them would overcome to act as a second wlan device.

Also - the wifi management tools in Raspberry are just horrible...

The best thing would be some tutorial that starts with a fresh Raspi3 or 4, say which wifi usb stick is used and how it was set up and what files have to be changed in what way to make it work.

In my case the software and Raspi do not really play together or at least i cannot find a way to make it work :-/

githlp, 2021-01-28:
Did it.

The problem was sitting in front of the monitor. The USB Dongle was not set up correctly, although it seemed to work properly.

Started over again right from the start. 

Now its working with wifi only and capturing. Thanks again! :-)

closed 2021-01-28 by githlp