KasperskyLab/TinyCheck #42: new source for malware IOC

athecat, 2021-02-01:
Hi
what about including/collaborating with Abuse.ch (https://abuse.ch/#projects)?

(thanks for your work)
A

felixaime, 2021-02-01:
Hello AtheCat, 

Thanks for your message! I'm gonna to think about that, why not if they are following mobile implants. After let's see if some ppl are also using our project to analyze their laptops etc. Can be more than good in that case!

Have a good day,
Félix.

athecat, 2021-02-01:
most of them are also C&C, so they could also abuse the mobile device downloading stalkerware.

If you could enable these IOC, I think for sure I'll begin analyzing laptops too :)
thanks
A

XlllllllX, 2021-07-23:
Hi!
I already analyze both "compu-phone" and laptop with TC. These IOC could improve TC efficiency imho.
Awesome tool, many many thanks for your job :)