KasperskyLab/TinyCheck #51: TinyCheck doesn't detect DNS queries to stalkerwares servers

leandreSL, 2021-02-16:
Hi,

After tested TinyCheck on a virtual environment (with a little bit of modifications in the code), i decided to test it on the purposed environment : a Raspberry Pi 4 (with a touchscreen).

But after fews tests with a stalkerware (in this test Snoopza) installed on a smartphone, i saw that it didn't detect first DNS query to ``` api.snoopza.com ```  ( it is easily viewable on the capture.pcap with a ```dns``` filter). 

On the virtual environment, I modified the ```zeekengine.py``` file to also loop on all dns queries and compare it with IOCs : 
```python
for d in self.dns:
                # Check for blacklisted domain.
                for domain in bl_domains:
                    if d["domain"].endswith(domain[0]):
                           .....
```

I wonder if i am doing a thing wrong, or if the analysis needs to loop also on DNS queries ? 

Thanks for the work on this amazing tool,
Léandre

felixaime, 2021-02-16:
Hello Léandre, 

Nice catch I have the same issue there, it put the domain `api.snoopza.com` in whitelisted coms in the report, even if that domain, nor its CIDR is whitelisted. I'm gonna trace all the thing and see where is the bug and come back to you this afternoon!

Félix.

felixaime, 2021-02-16:
Re Léandre,

Ok, its not related to the engine but a typo, the whitelisted CIDR : **199.36.154.0/2** -_-. Sure that if we whitelist all the internet, and not the CIDR 199.36.154.**0/23** it will not work. I'm gonna update the whitelist.json and also the update script to remove the entry. You can remove it by yourself by going to the backend, whitelist, search element and typing  **199.36.154.0/2**, after that, delete bouton. 

Have a good day,
Félix.

leandreSL, 2021-02-16:
Thank you for the quick reply,

It fixed everything, thank you !

I think i can close this issue.

Have a nice day too,
Léandre.

closed 2021-02-16 by leandreSL