KasperskyLab/TinyCheck #55: No internet connection, TinyCheck blocked on the QR page.

simone-s, 2021-02-20:
Dear,

I have installed a fresh TinyChek on a RPi3B+ with the last Rasberry OS Buster.
My configuration has:

- wlan0 onboard the RPi for the Internet connection
- wlan1 to work as AP 

Currently, I would create a portable configuration so, wlan0 accesses the Internet through tethering with iPhone(1).
Whereas, wlan1 should create the AP to investigate another iPhone, let's call this iPhone(2).

I'm able to access the backend and frontend.
I'm able to create the AP and generate the QR code.
But, when I try to connect the iPhone(2) to wlan1, TinyCheck doesn't release any private IP to the iPhone(2) and after a few on the iPhone appear the sentence "No Internet connection".
Moreover, TinyCheck doesn't go ahead it is stopped on the QR page.

Any idea to solve this issue?
Thanks.

felixaime, 2021-02-22:
Hello Simone!

So if I understand well, you are able to connect your iPhone(2) to the generated Wi-Fi network but you don't have any IP address released. Are you redirected to the page "Intercepting the communications of iPhone(2)" ? The issue that you describe can be an issue related to dhcpcd. Please look under `/etc/dhcpcd.conf` if you have these lines at the bottom:

```
## TinyCheck configuration ##

interface wlan1
   static ip_address=192.168.100.1/24
   nohook wpa_supplicant
```

And if the service is running or have issues by looking at `service dhcpcd status`.

Have a good day,
Félix.

simone-s, 2021-02-22:
Hello Félix,

you understood correctly.
The file /etc/dhcpcd.conf is correct. I mean it already reports the same lines that you have just posted.
The service si active and running.

Unfortunately, it is blocked on the QR code and it doesn't go ahead.

Thanks for the support.

felixaime, 2021-02-22:
Ok, looks like more hostapd issue so. Can you show your `/etc/hostapd.log` ?

simone-s, 2021-02-22:
Hello,
I don't have such a file in /etc/ folder.
What can I check?

felixaime, 2021-02-23:
Sorry, my bad, `/tmp/hostapd.log` (when it have started an AP/QRCode)

simone-s, 2021-02-24:
Hello Félix,
here you are the content of /tmp/hostpad.conf
`
Configuration file: /tmp/hostapd.log
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan1: Could not connect to kernel driver
Using interface wlan1 with hwaddr 86:b8:72:f5:42:01 and ssid "pi-3746"
wlan1: interface state COUNTRY_UPDATE->ENABLED
wlan1: AP-ENABLED 
wlan1: STA 72:fe:f9:e1:11:1a IEEE 802.11: associated
wlan1: AP-STA-CONNECTED 72:fe:f9:e1:11:1a
wlan1: STA 72:fe:f9:e1:11:1a RADIUS: starting accounting session 5DA5357C7AAABECE
wlan1: STA 72:fe:f9:e1:11:1a WPA: pairwise key handshake completed (RSN)
`
I add also the content of /tmp/hostpad.conf
`
country_code=GB
interface=wlan1
ssid=pi-3746
hw_mode=g
channel=7
auth_algs=1
wpa=2
wpa_passphrase=7a8bb4cf
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
disassoc_low_ack=0
`
Reading these two files, I noted that the access point is configured for a wrong country. I'm not in GB...

thanks.

felixaime, 2021-02-24:
Hello,

Ok, everything seems normal at that point. Its very strange. 

Can you check, once your phone connected to the AP, that you have something in the `/var/lib/misc/dnsmasq.leases` file ? TinyCheck tcheck the content of that file in order to redirect the user to the interception page. Maybe there is something wrong there...

Sorry for the response delay.
Félix.

simone-s, 2021-02-24:
Hello,

the file dnsmasq.leases is empty.
And the phone has an IP 169.254.....

The strange things is that it is a fresh installation. I mean I downloaded TinyCheck from Github and installed by following the instruciotns.
I checked also the `iw list` command if I had problems with the WiFi dongle and the output is 1 with one dongle and 2 with another one.

Any other idea?

Regards.

simone-s, 2021-02-24:
Hello,

I don't know if it might be helpful, but if run `dnsmasq service `I got

> XX@tinycheck:~ $ sudo service dnsmasq start
> Failed to start dnsmasq.service: Unit dnsmasq.service not found.

Does Tinycheck need some dependenices? Becacuse I installed a fresh OS Buster on RPi and then Tinycheck. Nothing else.

Regards.

felixaime, 2021-02-25:
Strange because dnsmasq is normally installed by the installation script. You can try to reinstall it with the command : 

`sudo apt-get install dnsmasq`

simone-s, 2021-02-25:
Hello,

i have installed the **dnsmasq** and now the service is active and running.
Unfortunately, the file `/var/lib/misc/dnsmasq.leases` is empty, i.e., 0 byte.

What I can try to get it work?

thanks!

felixaime, 2021-02-26:
Ok, still strange. Have you in your `/etc/dnsmasq.conf` these lines ?

```
## TinyCheck configuration ##

interface=wlan1
dhcp-range=192.168.100.2,192.168.100.3,255.255.255.0,24h
```

simone-s, 2021-02-26:
Hello,

we did a step.
I hadn't the lines that you indicated. It's very strange, because I guessed that the installation adds these lines. What could be the problem?
 
Once I added those in `dnsmasq.conf` I was able to perform the first capture... thanks!
Unfortunately, it didn't solve the original problem.
Indeed, Tinycheck, is using the local WLAN in my home and not the iPhone tethering as I indicated in my first post.
If I connect wlan0 to Internet through the iPhone tethering, TinyCheck gets IP in the range 172.0... and it is not able to create the AP. And I'm not able to connect the second mobile phone that I would analyze.

Thanks.

felixaime, 2021-02-26:
> Hello,
> 
> we did a step.
> I hadn't the lines that you indicated. It's very strange, because I guessed that the installation adds these lines. What could be the problem?
> 

Yeah, honestly, I don't know, I don't understand why `dnsmasq` haven't been installed and the install script leveraged no errors regarding that. I need to investigate. Maybe can be good for you to uninstall it and redo the install process, or just update it. 

> Unfortunately, it didn't solve the original problem.
> Indeed, Tinycheck, is using the local WLAN in my home and not the iPhone tethering as I indicated in my first post.

Ok, so for that you need to edit `/etc/wpa_supplicant/wpa_supplicant.conf` and delete your home network. Your question bring me some bug as there is no network priority in the generated `wpa_supplicant.conf`. I need to fix that.

> If I connect wlan0 to Internet through the iPhone tethering, TinyCheck gets IP in the range 172.0... and it is not able to create the AP. And I'm not able to connect the second mobile phone that I would analyze.
> 
> Thanks.

That more than strange, its 172.0.XXX.XXX or 172.16.XXX.XXX ? Can you try to connect your Rpi to your iPhone by the usual Wifi Manager and try to browse internet ?

simone-s, 2021-02-26:
Hello Fèlix,

thanks removing the home network it works as portable setup.

Two things:

1.  ok I can update it, running the update.sh script? Does it remove all the configurations that we discussed in this thread?
2.  I'm newbie with Tinycheck, do I have to install particular IOC  to check a mobile phone? I mean where I can find things to investigate in deep the phone?

Thanks for your support!

felixaime, 2021-02-26:
Regarding the update it will not delete your config and the database, just update the code. So no config file deleted (dnsmasq and dhcpcd.conf will remain unchanged.)

For the IOCs, they are automatically updated by the watchers, normally, by browsing the backend, you can check for IOCs, for example by search *.com in the IOC search field.

simone-s, 2021-02-26:
> Regarding the update it will not delete your config and the database, just update the code. So no config file deleted >(dnsmasq and dhcpcd.conf will remain unchanged.)

ok thanks!
 
> For the IOCs, they are automatically updated by the watchers, normally, by browsing the backend, you can check for IOCs, >for example by search *.com in the IOC search field.

I'm sorry but I don't understand.  What should I find by searching *.com?

felixaime, 2021-02-26:
Hum, just to see if there is IOCs in the database :)

simone-s, 2021-02-26:
> Hum, just to see if there is IOCs in the database :)

Ok, looking at the Tinycheck backend, I understood that there are some IOC included.
But, can I add more? I mean can I search for more IOC on the Internet, and if yes, where?
Or only IOC that you tested and approved can be imported into the Tinycheck?

felixaime, 2021-02-26:
Regarding the IOCs, you can add any IOCs that you want (Domains, IP addresses etc.), even legit stuff. As its your own tinycheck instance, you can do what you want. To add them, you can look at this page to see how to do that: 

https://github.com/KasperskyLab/TinyCheck/wiki/TinyCheck-backend

Regarding to the Stalkerware IOCs, we try to maintain a good list of them. At each TinyCheck system startup the IOCs are updated.

simone-s, 2021-02-26:
> Regarding the IOCs, you can add any IOCs that you want (Domains, IP addresses etc.), even legit stuff. As its your own tinycheck instance, you can do what you want. To add them, you can look at this [...]

Ok, thanks! I'll try it!

I think that we can now close this thread because the problem has been solved.
Maybe a suggestion, if there are some needed packages we can run an apt-get install <dependencies> during the installation.

Regards.

felixaime, 2021-02-26:
> needed packages we can run an apt-get install during the installation.

Yeah, but its what we already did :)

```bash
if [[ $1 == "dnsmasq" || $1 == "hostapd" || $1 == "tshark" || $1 == "sqlite3" || $1 == "suricata"  || $1 == "unclutter" ]]; then
       apt-get install $1 -y
```

I keep it open, just to remind me to solve the issue with the networks priority :)
Have a good weekend!

Félix.

closed 2021-03-25 by felixaime