KasperskyLab/TinyCheck #68: Do you have a plan for time-variant events?

bl4ckh0l3z, 2021-04-08:
Going straight to the point: is there any chance to raise alerts by feeding on snort rules like this?

```
alert TCP $EXTERNAL_NET any -> $HOME_NET any (msg:"This is just an example of time-variant rule"; flags:S; threshold: type threshold, track by_dst, count 1000 ,
seconds 60; sid: 5000002;)
```

Count is accrued over a specific period of time, but at the moment time-variant analysis is totally ignored by tinycheck...and, as far as I know, snort rules are evaluated in a shoot via suricata...isn’t true?

Thanks.

felixaime, 2021-04-09:
Hello bl4ckh0l3z,

Yeah, strange that it is completely ignored. I'm gonna work on that to see what's wrong because can be usefull to detect some tunnels to legit services (telegram, dropbox etc etc.)  and yes, the rules are evaluated via by using suricata!

bl4ckh0l3z, 2021-04-09:
Unfortunately, it's the expected behaviour when suricata/snort are used to analyze a pcap file capture ex-post.

A workaround could be tcpreplay capture.pcap on the loopback, by taking into account the appropriate pps rate, and start snort/suricata listening on the loopback interface.

Event occurrences will be accrued correctly and alert will be triggered accordingly.

This approach is neither elegant, nor expensive to implement...most of the effort is duplicate suricataengine.py to snortengine.py and install snort too. 

Btw, I'm pretty sure you will spot a more appropriate solution...

Ps. Extra-bonus: if you try to import a snort rule like
```
[...] pcre: "/^\x17\x03.../" [...]
```
every occurrence of "\x" will be changed to "/x" from the BE and this error propagates to the SQLite-db...resulting in no match for the rule. Cheers