KasperskyLab/TinyCheck #76: CERT-FR's IOC

XlllllllX, 2021-07-23:
Hi again!

I try to add IOC from cert-fr without success. I tried:
- c/p raw json
- import json file
In all cases, I get
`✗ 303 IOCs not imported, see details below.`
with
```
[…]
&#34;galaxy&#34;:,                                   | Wrong IOC format
&#34;shadowattribute&#34;:,                          | Wrong IOC format
&#34;tag&#34;:,                                      | Wrong IOC format
&#34;category&#34;:&#34;network activity&#34;,       | Wrong IOC format
&#34;comment&#34;:&#34;adresses ip reliu00e9es&#34;, | Wrong IOC format
&#34;deleted&#34;:false,                             | Wrong IOC format
&#34;disable_correlation&#34;:false,                 | Wrong IOC format
[…]
```
I suppose it's parsing related. Can I do something to get it work?
(for example a script to preprocessing the file before importing it in TC)

'good'day!

felixaime, 2021-08-05:
Hello,

Yeah, its possible that they are MISP or whatever formatted. You can try with jq : 

`curl -s https://www.cert.ssi.gouv.fr/uploads/CERTFR-2021-IOC-003-IOC.json 2>&1 | jq -r '.Event.Attribute[].value'`

And then, copy paste the result in the backend.

Félix.

XlllllllX, 2021-08-12:
Many thanks, Félix! :)