KasperskyLab/TinyCheck #78: not saving PDF report

harut420, 2021-08-18:
Hello,
is there anything i should enable to save PDF report?

inside the USB key i have under TinyCheck dir only : 
capture.pcap
assets  ( dir) 

i do not see any pdf here.
Thanks

harut420, 2021-08-19:
running :
sudo python3 /usr/share/tinycheck/analysis/analysis.py /tmp/63FF3B7D/

i get error on libpango : 

File "/usr/share/tinycheck/analysis/analysis.py", line 71, in <module>
    report.generate_report()
  File "/usr/share/tinycheck/analysis/classes/report.py", line 63, in generate_report
    htmldoc = HTML(string=content, base_url="").write_pdf()
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/__init__.py", line 183, in write_pdf
    counter_style=counter_style, image_cache=image_cache)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/__init__.py", line 136, in render
    optimize_size, font_config, counter_style, image_cache)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/document.py", line 885, in _render
    [Page(page_box) for page_box in page_boxes],
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/document.py", line 885, in <listcomp>
    [Page(page_box) for page_box in page_boxes],
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/__init__.py", line 124, in layout_document
    pages = list(make_all_pages(context, root_box, html, pages))
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/pages.py", line 802, in make_all_pages
    page, resume_at = remake_page(i, context, root_box, html)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/pages.py", line 741, in remake_page
    page_number, page_state)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/pages.py", line 552, in make_page
    positioned_boxes, adjoining_margins, discard=False)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 60, in block_level_layout
    page_is_empty, absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 75, in block_level_layout_switch
    discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 128, in block_box_layout
    absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 520, in block_container_layout
    absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 60, in block_level_layout
    page_is_empty, absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 75, in block_level_layout_switch
    discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 128, in block_box_layout
    absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 520, in block_container_layout
    absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 60, in block_level_layout
    page_is_empty, absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 75, in block_level_layout_switch
    discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 128, in block_box_layout
    absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 520, in block_container_layout
    absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 60, in block_level_layout
    page_is_empty, absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 75, in block_level_layout_switch
    discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 128, in block_box_layout
    absolute_boxes, fixed_boxes, adjoining_margins, discard)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/blocks.py", line 379, in block_container_layout
    for i, (line, resume_at) in enumerate(lines_iterator):
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/inlines.py", line 50, in iter_line_boxes
    absolute_boxes, fixed_boxes, first_letter_style)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/inlines.py", line 178, in get_next_linebox
    line_children=[])
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/inlines.py", line 848, in split_inline_box
    line_placeholders, child_waiting_floats, line_children))
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/inlines.py", line 706, in split_inline_level
    line_children)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/inlines.py", line 848, in split_inline_box
    line_placeholders, child_waiting_floats, line_children))
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/inlines.py", line 706, in split_inline_level
    line_children)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/inlines.py", line 848, in split_inline_box
    line_placeholders, child_waiting_floats, line_children))
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/inlines.py", line 683, in split_inline_level
    context, box, max_x - position_x, skip)
  File "/usr/local/lib/python3.7/dist-packages/weasyprint/layout/inlines.py", line 1104, in split_text_box
[... cut ...]
  File "/usr/local/lib/python3.7/dist-packages/cffi/api.py", line 912, in __getattr__
    make_accessor(name)
  File "/usr/local/lib/python3.7/dist-packages/cffi/api.py", line 908, in make_accessor
    accessors[name](name)
  File "/usr/local/lib/python3.7/dist-packages/cffi/api.py", line 838, in accessor_function
    value = backendlib.load_function(BType, name)
AttributeError: function/symbol 'pango_context_set_round_glyph_positions' not found in library 'libpango-1.0.so.0': /lib/arm-linux-gnueabihf/libpango-1.0.so.0: undefined symbol: pango_context_set_round_glyph_positions

aostano, 2021-08-29:
Hi harut420,
you need to upgrade with "apt-get upgrade" followed by "pip3 install -U django-weasyprint".

Greetings

athecat, 2021-09-03:
Hi
I'm having the same issue on a completely fresh installed Raspberry pi$4.
Everything works in installation, updates, acquisition and analysis, but in the downloded file the pdf is missing.
I've tried what @aostano suggested, but it didn't solve the problem.

Thanks for your help

harut420, 2021-09-03:
Hi ,
it worked for me.
Harut420

aostano, 2021-09-03:
Errata corrige:
logged in as pi user, you need to upgrade with "sudo apt-get upgrade" followed by "sudo pip3 install -U django-weasyprint".

Greetings

jojo018, 2021-11-10:
@aostano I tried to fix the pdf issue with the recommended commands, but it didn't work. There's still no pdf report in the zip file. 

I installed tinycheck on raspberry pi 3 and I have to admit that I didn't work with raspberry pi before

HonourableTyr, 2022-02-21:
Same problem for me and 

_sudo apt-get upgrade_

followed by

_sudo pip3 install -U django-weasyprint_

does not work. The package(s) failed to install initially. I rebooted and tried again ant it seems to have installed.

However, still no PDF report.

athecat, 2022-02-22:
In my case the issue was caused by me not installing "Raspbian Buster OS 32bit full" (note the FULL!) 
Using the right OS, I've managed installing and using TinyCheck with no more issue.

HonourableTyr, 2022-02-22:
> In my case the issue was caused by me not installing "Raspbian Buster OS 32bit full" (note the FULL!) Using the right OS, I've managed installing and using TinyCheck with no more issue.

This isn't particularly helpful given that Pi OS Buster (Legacy) is no longer available as a full install (with recommend software). It is only available as "Lite" or "With Desktop". I'm using the January 2022 version of Buster, again the only version available.

Considering that Buster is considered a legacy OS it would be great for TinyCheck to be updated to support Bullseye.

HonourableTyr, 2022-04-27:
> The weasyprint version can be downgraded by the following command: `sudo pip3 install weasyprint==52.5`
> 
> django-weasyprint complains that dependencies are not met, because weasyprint >= 53.0 are required but reporting works fine afterwards.

Awesome, thanks! I've got everything working now. Next step is full testing on seeded devices. Then I can move on to user testing, to make sure that when in self service mode users don't have any difficulties.

EvgenyAblesov, 2023-08-29:
Hello everyone in this thread! 

We did a lot of work to improve TinyCheck installation process including dependencies resolution. We believe that this problem is already solved.

Have a good day!