KasperskyLab/TinyCheck #90: fresh install fails (some debugging inside)

fosBHY, 2021-12-30:
Inspired by the talk at rc3 about this project I wanted to try it out.

after a fresh installation on raspbian buster (tried bullseye where at least the backend and frontend ran until meeting import errors, remembered the speaker at rc3 said buster would work better) the thing did nothing. login shows:

```
systemctl status tinycheck-watchers:
    ...
    ModuleNotFoundError: No module named 'sqlalchemy'

systemctl status tinycheck-frontend:
    ...
    ModuleNotFoundError: No module named 'flask'

systemctl status tinycheck-backend:
    ...
    ModuleNotFoundError: No module named 'flask'
```

apparently line `320` might not have worked?

```
 python3 -m pip install -r "$SCRIPT_PATH/assets/requirements.txt"
```

My try `python3 -m pip install -r requirements.txt` revealed an error (multiple similar all with cryptography package)
```
  Failed building wheel for cryptography
```

debug assistance of pip(?) recommends upgrade.

```
python3 -m pip install --upgrade pip
```

and again, somewhat more compact:
```
This package requires Rust >=1.41.0.
----------------------------------------
ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects
```

So this might have to do with `cryptography` from version 3.5 onwards relies on rust >=1.45.
https://cryptography.io/en/latest/installation/

edit `requirements.txt`
```
cryptography < 3.49
...

which made the install ( `python3 -m pip install -r requirements.txt` ) go through

fosBHY, 2021-12-30:
now the web backend and frontend work-

still, like in the bullseye test I did earlier it is stuck at creating wifi network forever ( `/generate-ap`  url )

fosBHY, 2021-12-30:
ok. some more info. as you write in https://github.com/KasperskyLab/TinyCheck/wiki/Timeout-on-the-web-interface you are happy about bugreports.

when debugging and starting the frontend myself the error now is:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/qrcode/image/pil.py", line 5, in <module>                                                                                     
    from PIL import Image, ImageDraw
  File "/usr/local/lib/python3.7/dist-packages/PIL/Image.py", line 114, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2091, in __call__
    return self.wsgi_app(environ, start_response)
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2076, in wsgi_app
    response = self.handle_exception(e)
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1518, in full_dispatch_request                                                                            
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1516, in full_dispatch_request                                                                            
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1502, in dispatch_request                                                                                 
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/share/tinycheck/server/frontend/app/blueprints/network.py", line 44, in api_start_ap                                                                            
    return jsonify(network.start_ap())
  File "/usr/share/tinycheck/server/frontend/app/classes/network.py", line 189, in start_ap                                                                                  
    "qrcode": self.generate_qr_code()}
  File "/usr/share/tinycheck/server/frontend/app/classes/network.py", line 205, in generate_qr_code                                                                          
    self.AP_SSID, self.AP_PASS))
  File "/usr/local/lib/python3.7/dist-packages/qrcode/main.py", line 13, in make
    return qr.make_image()
  File "/usr/local/lib/python3.7/dist-packages/qrcode/main.py", line 296, in make_image
    from qrcode.image.pil import PilImage
  File "/usr/local/lib/python3.7/dist-packages/qrcode/image/pil.py", line 7, in <module>                                                                                     
    import Image

ModuleNotFoundError: No module named 'Image'
```

3frogx, 2021-12-30:
I’m also inspired by the r3c-talk.
On my raspi3 I startet as follow:

clean install image: Raspberry Pi OS (Legacy) from 2021-12-02 (buster) with Imager
over ssh, do the installation as described here on github as root

kioskmode doesn't work immediately because tinycheck-kiosk was looking for chromium-browser, I only have chromium, I set a symbolic link.

Now everything is working except the PDF Export. I tried the commands as described in #78 but with no success.
Always get: function/symbol 'pango_context_set_round_glyph_positions' not found in library 'libpango-1.0.so.0': /lib/arm-linux-gnueabihf/libpango-1.0.so.0: undefined symbol: pango_context_set_round_glyph_positions

cyb3rwr3ck, 2022-01-02:
@fosBHY which hardware are you using? I have a zero w available for some testing (buster image) and can confirm the problem with the cryptography packet which you've fixed. Additionally, on the zero w, which is armv6 hardware, the nodejs installation is incompatible and returns an error because it detects the armv6 hardware ("Message: _## You appear to be running on ARMv6 hardware. Unfortunately this is not currently supported by the NodeSource Linux distributions. Please use the 'linux-armv6l' binary tarballs available directly from nodejs.org for Node.js 4 and later._"). I've installed v11.15.0 manually from nodejs.org, which seems to be the latest tarball with v6 support and now at least it starts to build. But it breaks  with `Error: @vitejs/plugin-vue requires vue (>=3.2.13) or @vue/compiler-sfc to be present in the dependency tree`. No time left to investigate any furhter for now...

From my point of view the installer is broken in several ways depending on different circumstances like hw-versions etc. For non technical people it seems pretty hard to get around that and I would suppose that creating and hosting an raspi image would make much more sense.

fosBHY, 2022-01-09:
I used a raspberry pi 3b+

I second your view on the installer and also think a dedicated Image would be much easier for people to actually use, even nonsavvy

unmerkbar, 2022-01-09:
Hi there.

Wanted to join the discussion. Same for me, inspired by the rC3 Talk...

I have an Raspberry Pi 4 Model B Rev 1.2 with a new Image of Buster by the Raspi Imager and did the installation as described in the wiki. No errors during installation. I access the front-end via browser within the home network. It stucks at the same point when generating the AP on the /generate-ap url. Enabling the debug mode showed the same ModuleNotFoundError as described in https://github.com/KasperskyLab/TinyCheck/issues/90#issuecomment-1002843170 by @fosBHY.

Dmourn, 2022-02-03:
I had many issues trying to install this on the newest version of rpi os (debian 11).

The install script needed a little love (maybe needs a little more).
I also used a venv for python and and systemd. 

If anybody would like to help test it I'd appreciate it.

EvgenyAblesov, 2023-08-17:
Hello to everyone!

We did some work on installer script recently. 
For now it runs with no issues (almost :-) ).

Any issues with versions and deps were fixed also, so for now it seems like this issue can be closed.

closed 2023-08-17 by EvgenyAblesov