KasperskyLab/TinyCheck #93: tinycheck-frontend.service failed

jgee090, 2022-02-14:
tinycheck-frontend.service failed, not able to get into http://tinycheck.local or http://127.0.0.1
Any ideas? Can sen logs/files if needed

● tinycheck-frontend.service - TinyCheck frontend service
   Loaded: loaded (/lib/systemd/system/tinycheck-frontend.service; enabled; vend
   Active: failed (Result: exit-code) since Mon 2022-02-14 16:25:27 GMT; 17s ago
  Process: 400 ExecStart=/usr/bin/python3 /usr/share/tinycheck/server/frontend/m
 Main PID: 400 (code=exited, status=1/FAILURE)

Feb 14 16:25:27 tinycheck python3[400]:     from pyudev.core import Context, Enu
Feb 14 16:25:27 tinycheck python3[400]:   File "/usr/local/lib/python3.7/dist-pa
Feb 14 16:25:27 tinycheck python3[400]:     from pyudev.device import Devices
Feb 14 16:25:27 tinycheck python3[400]:   File "/usr/local/lib/python3.7/dist-pa
Feb 14 16:25:27 tinycheck python3[400]:     from ._device import Attributes, Dev
Feb 14 16:25:27 tinycheck python3[400]:   File "/usr/local/lib/python3.7/dist-pa
Feb 14 16:25:27 tinycheck python3[400]:     from six.moves import collections_ab
Feb 14 16:25:27 tinycheck python3[400]: ImportError: cannot import name 'collect
Feb 14 16:25:27 tinycheck systemd[1]: tinycheck-frontend.service: Main process e
Feb 14 16:25:27 tinycheck systemd[1]: tinycheck-frontend.service: Failed with re
lines 1-16/16 (END)

PlazzmiK, 2022-02-16:
This seems to fix it:
`sudo python3 -m install --upgrade six`

jgee090, 2022-02-16:
I tried that, tried version 1.11,1.12,1.15 and upgraded to 1.16 still not working
Running python3 3.7.3 and python 2.7.16

PlazzmiK, 2022-02-16:
> I tried that, tried version 1.11,1.12,1.15 and upgraded to 1.16 still not working Running python3 3.7.3 and python 2.7.16

Did you reinstall TinyCheck after install/upgrading six?

jgee090, 2022-02-17:
Got it working on Debian Bullseye rather than Buster

closed 2022-02-17 by jgee090