KasperskyLab/TinyCheck #95: Unable to launch/access frontend.

HonourableTyr, 2022-02-17:
I have been having numerous issues trying to get TinyCheck to run and slowly overcoming each one. The final hurdle seems to be getting the front end to work.

I am able to access the backend at https://127.0.0.1 and configure it easily enough. It detects the two WiFi devices, onboard and USB, as well as the disconnected ethernet. The onboard WiFi is connected to a router and the internet.

Whenever I try to launch or access the frontend I just get an error message in Chromium as below:

---
### This site can’t be reached
**127.0.0.1** refused to connect.
Try:
- Checking the connection
- [Checking the proxy and the firewall](chrome-error://chromewebdata/#buttons)

ERR_CONNECTION_REFUSED

---

I am still a novice when it comes to Linux and anything Raspberry Pi.

I am running TinyCheck on an Raspberry Pi 400 with Raspbian 10 "Buster". It is a fresh standard 32-bit image with all packages updated.

No errors were reported during installation of TinyCheck.

I did have to make a minor modification the execution script as "chromium-browser" doesn't exist on the current Raspbian install, it is just called "chromium".

If i bash the kiosk script in the terminal I get the following errors:

---

_Opening in existing browser session.
[3187:3187:0217/142107.370387:ERROR:broker_posix.cc(43)] Invalid node channel message
[3184:3184:0100/000000.158348:ERROR:gpu_init.cc(426)] Passthrough is not supported, GL is desktop
[3184:3184:0100/000000.206342:ERROR:broker_posix.cc(43)] Invalid node channel message_

---

That doesn't mean much to my non-programmer mind and I'd appreciate some help.

Dmourn, 2022-02-18:
I was unable to reproduce your problem. I didn't have to change the the script either, are you using the buster version marked (Legacy)?

I'm also using ethernet, can you try that to see if anything is different?

Have you tried connecting to the pi with a different device on the network your router is on?

HonourableTyr, 2022-02-18:
Yes, I am using the OS marked "legacy" because the requirements stated "Buster". There is no current version of Pi OS Buster, based off Debian 10. The current version is code named "Bullseye", based off Debian 11.

Is the wiki incorrect and I should be using Bullseye?

I'll try both ethernet and connecting from another device.

PlazzmiK, 2022-02-18:
`sudo python3 -m install --upgrade six`

Before installing, or followed by a reïnstall of tinycheck. Pull request #94 seems to fix this.

HonourableTyr, 2022-02-18:
> `sudo python3 -m install --upgrade six`
> 
> Before installing, or followed by a reïnstall of tinycheck. Pull request #94 seems to fix this.

That just returns an error:

"No module named install"

HonourableTyr, 2022-02-18:
> I was unable to reproduce your problem. I didn't have to change the the script either, are you using the buster version marked (Legacy)?
>
> I'm also using ethernet, can you try that to see if anything [...]

So I have tried the following on both Buster and Bullseye connected via ethernet the result is almost the same.

Running the desktop shortcut which bashes kiosk.sh does nothing whatsoever without editing the contents of kiosk.sh. This is because, "chromoim-browser" doesn't exist and it returns the error as below:

_$ bash /usr/share/tinycheck/kiosk.sh
  kiosk_mode: true
$ /usr/share/tinycheck/kiosk.sh: line 18: /usr/bin/chromium-browser: No such file or directory
bash /^Cr/share/tinycheck/kiosk.sh_

Editing it to just read "chromium" launches chromium in full screen for kiosk mode. This returns the web error below as per my original results.

---

### This site can’t be reached
**127.0.0.1** refused to connect.
Try:

Checking the connection
Checking the proxy and the firewall
ERR_CONNECTION_REFUSED

---

I am no longer getting any other errors bashing kiosk.sh in the terminal though, only the error above.

_$ bash /usr/share/tinycheck/kiosk.sh
  kiosk_mode: true
$ Opening in existing browser session._

zippelzapp, 2022-02-19:
Had the same problem. PlazzmiKs Solution works. But i had to use this before installation of tinychek: sudo python3 -m pip install --upgrade six

HonourableTyr, 2022-02-19:
> Had the same problem. PlazzmiKs Solution works. But i had to use this before installation of tinychek: sudo python3 -m pip install --upgrade six

Thanks, I already tried that and the command did not work as per my reply to PlazzmiK.

zippelzapp, 2022-02-19:
Okay, sorry. I thought it might help you, because i had the same error "No module named install". So i used "pip install" instead of install only and the installation was succesful.

HonourableTyr, 2022-02-19:
> Okay, sorry. I thought it might help you, because i had the same error "No module named install". So i used "pip install" instead of install only and the installation was succesful.

Ah, my bad I misread your post and complete missed the 'pip' added. I'll try that shortly!

HonourableTyr, 2022-02-21:
> Okay, sorry. I thought it might help you, because i had the same error "No module named install". So i used "pip install" instead of install only and the installation was succesful.

Problem solved! AFAIK.

Testing capture. Although doesn't appear to be bridging to the internet but that is likely just a configuration issue.

HonourableTyr, 2022-02-21:
Fully resolved! I was inadvertently using Raspbian 11 in my last test instead of 10 (Buster). I must have grabbed the wrong SD card (I've got 7 with different OS setups).

The shortcut on the desktop doesn't work nomatter what but I can just use the IP address to access the front and back ends.

Thanks to zippelzapp for the solution. The rest was user error (as is usually the case).