KasperskyLab/TinyCheck #97: Install hangs on 'Feeding your TinyCheck... fresh IOCs and whitelist'

nainsworthrv, 2022-03-06:
Hi all,
Running Rapsberry Pi OS (Legacy) Buster v10.11

Attempted to install fresh version of TinyCheck - second time trying and installation hung continuously on "Feeding your Tinycheck instance with fresh IOCs and whitelist, please wait'.

I waited overnight but it did not proceed. Tried again today and same result.
I rebooted it and tried to the result anyway - but could not get TinyCheck frontend working --- error connection refused. Might have been the same as the problem with the Python 'six' package that is mentioned in issue #95 though looking back on it.

Any workarounds for this problem with the updates?

Thanks!

sillo1991, 2022-04-22:
I have the same problem

wangyanwen, 2022-07-11:
I have the same problem,and my device is pi 400.

Raschi1, 2022-07-21:
Had the same problem on Pi4B. I tried use Bullseye but TinyCheck installation refused to start on unsupported Debian version, so Buster is the only option.

So I returned to Buster and prior to installation of TinyCheck I upgraded python six package (as recommended for another issue)
`sudo python3 -m pip install --upgrade six`
and manually installed chromium browser
`sudo apt install chromium-browser`.
TinyCheck installation then still hung on "Feeding your Tinycheck instance with fresh IOCs and whitelist", but after breaking with Ctrl+C system was restarted and TinyCheck kiosk started normally. When looking to backend, IOCs are present (3997 of them)

mrludo001, 2022-07-21:
I suspect you might have tried with a 64-bit version of Bullseye.  The
installer script is looking for an identifing entry that is different on
the 64-bit Bullseye to that on the 32-bit Bullseye.  I will look tomorrow
and tell you what it is precisely...

-- Dave

mrludo001, 2022-07-22:
The installer script is looking for an identifying entry that is different on
the 64-bit Bullseye to that on the 32-bit Bullseye.

The install.sh script uses _grep_ to look inside **/etc/os-release** for the string **ID_LIKE=debian**

On the **64-bit** Bullseye image, **/etc/os-release** does not have this string in it.
The file contains:

**ID=debian**

On the **32-bit** Bullseye image, **/etc/os-release** does have the string.
The file contains:

**ID=raspian
ID_LIKE=debian**

Raschi1, 2022-07-22:
You might be right, when I looked back to Pi imager, I really might have installed Bullseye 64bit. But I am not going to try it again with Bullseye 32bit, as the Buster is now running for me.

clanoftheducks, 2022-07-22:
Did the installation script manage to install zeek on your Buster 32bit ? It fails one mine: despite checking that the opensuse repository is in my sources, and `apt install`ing manually, `apt` does not want to get the package: 

```
pi@tinycheck:~ $ sudo apt update && sudo apt install zeek
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease                                    
Get:3 http://download.opensuse.org/repositories/security:/zeek/Raspbian_10  InRelease [1,534 B] 
Hit:4 https://deb.nodesource.com/node_14.x buster InRelease                                     
Fetched 1,534 B in 2s (1,006 B/s)                   
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package zeek is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'zeek' has no installation candidate
pi@tinycheck:~ $ uname -a
Linux tinycheck 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
```

Have you guys have trouble at that ? If not I might create a separate issue.

clhuby, 2022-07-29:
Zeek don't release Zeek5 binaries for Buster and instead only release Zeek4 (now called zeek-lts) which is why the TinyCheck script fails to install. You either have to install TinyCheck on Bullseye which has Zeek 5 binaries, or manually install the package "zeek-lts" on Buster before running the TinyCheck install script.

Zeek page showing only Bullseye (Debian 11) support https://software.opensuse.org//download.html?project=security%3Azeek&package=zeek

Zeek-LTS page showing Buster (Debian 10) support https://software.opensuse.org//download.html?project=security%3Azeek&package=zeek-lts

The install.sh needs to be modified to do "apt install zeek-lts" when it detects Buster to fix this bug.

mrludo001, 2022-08-10:
I added a comment onto the pull request Pr/92 #106 which has not yet been approved for merging. This Pr also addresses the issue of incorrect identification of Bullseye 64-bit release.

Hopefully the contributor of Pr/92 #106 will adjust the fix to accommodate this as well.