KasperskyLab/TinyCheck #98: Internet Bridge not working

WanderingCoder-Omen, 2022-04-11:
I have installed TinyCheck on a Raspberry Pi 4B with OS Buster with Desktop environment and security updates. wlan0 is connected to my home wi-fi and wlan1 is generating the AP for the phone to connect to. The ephemeral network is generated and phones can scan the QR code to join the network. However, the there's no internet in the generated network. The home wi-fi is fine and I even used eth0 in place of wlan0. But it doesn't work. Can you please help me out?

jgee090, 2022-05-13:
Had the same issue, this worked for me

`sudo apt install iptables-persistent -y`

fixpc82, 2022-09-05:
Hello everyone.  I made the changes you indicated but they continue to not connect

tinycheck correctly generates the Wi-Fi network and the IP is assigned within the indicated range.  but I still don't surf the internet with my mobile phone connected to the Wi-Fi network

you can help me?

lightkun-yagami, 2022-10-07:
I can verify that the bridge to the internet is working as of 10/7/2022. See the link that it is working: https://imgur.com/a/wap4oho

Malpaga, 2022-10-12:
Hello all,

I am facing the same problem, where everything works properly but network bridge doesn't.
Config has been done on a Raspberry Pi 3 Model B with OS Buster,
The Raspberry itself has internet access through eth0 interface.
Installation went smoothly, web interface is reached, ephemereal network is properly created and found by phone, but does not provide internet access.

Changes to the iptables rules did not fix the problem, any other clues ?

claudiopolis, 2022-10-16:
I am having the same issue on Buster (hardware is a Seeed Studio reTerminal, which has CM4 - really promising hardware for this project, as it comes with the touchscreen in a nice enclosure, all at a reasonable price given supply chain issues for regular RasPi boards). 

When I launch the ephemeral network and then SSH into the unit via the LAN interface, I can ping external hosts fine from the default source IP (which is the LAN port, which is plugged into upstream router), but when I specify the flag "-I 192.168.100.1" which is the IP of the ephemeral wlan, then no joy.

claudiopolis, 2022-10-16:
I re-imaged my device with Bullseye (32-bit) instead, and it all works great. Sadly, I didn't save more data from the problem on Buster. I noticed on Buster that before using the front-end to launch the ephemeral network, ifconfig showed only 3 interfaces, but afterwards, it showed 5 or 6. Bullseye still only shows 3 interfaces (eth0, l0 and wlan0) after launching the ephemeral, so I hope that's a clue for anyone who might not have the option to switch to Bullseye. 

FY & FWIWI, this is the 2nd problem I had with this solution that got fixed just by changing from Buster to Bullseye (the first being https://github.com/KasperskyLab/TinyCheck/issues/110).

Malpaga, 2022-10-19:
Fixed my problem by just adding a masquerade iptables rule :

`iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE`

Make sure the output interface is the one facing your router.

jeffman78, 2023-06-11:
I experienced the same issue (successfully connected to the ephemeral wifi - wlan1 - but no network connection to the internet).
After looking at the code, I discovered the IP range used for the ephemeral wifi is configured to used 192.168.100.0/24 and it was overlapping with the IP range allocated on wlan0. After updating my network config on a different IP range, everything worked like a charm.

closed 2023-08-17 by EvgenyAblesov