KasperskyLab/TinyCheck #101: How to Open the PDF Reports

Toaslaver, 2022-05-31:
Hi everybody, i have a problem I'm trying to open the capture.pcap file but i don't know which app i should use, any reccomendation? 
and also when i'm saving it in the USB pen, there absolutely no PDF report named report.pdf.
:/

Toaslaver, 2022-05-31:
i'm using the TinyCheck from a Raspberry pi4

felixaime, 2022-06-04:
For the capture.pcap, you can use Wireshark or NetworkMiner (https://www.netresec.com/?page=NetworkMiner) for softwares with a graphical Interface. 

So you have the capture.pcap but no report.pdf ?

Toaslaver, 2022-06-06:
Yes, I have the capture.pcap but no pdf report. I can open it only with
notes but it's not understandable.

HonourableTyr, 2022-07-17:
Hi, looks like you're having the same issue I was. Check my thread for the solution.

https://github.com/KasperskyLab/TinyCheck/issues/78#issuecomment-1109699133

Toaslaver, 2022-10-11:
so if i extract the file the results is this (results of extraction)
i'm trying to open the capture with wireshark but i'm not so familiar with
it.

EvgenyAblesov, 2023-08-29:
Hello everyone in this thread! 

We did a lot of work to improve TinyCheck installation process including dependencies resolution. We believe that this problem is already solved. So please try to reinstall TinyCheck.

Have a good day!

closed 2023-08-29 by EvgenyAblesov