KasperskyLab/TinyCheck #103: Raspberry alternative "Debian" TinyCheck install not working (Network issue: "You must select two interfaces, exiting")

Pitmit, 2022-06-22:
We tried different debian OEMs.
Kali Ubuntu anything even in Vm and on Hardware.
The OS always seems to accept the 2 network interfaces.
However during installation tinycheck doesnt select a 2nd interface and the read message (exciting)
We tried any possible hwardware VM situation we were capable of.

iw list doesnt show 2 interfaces but ifconfig -a shows them perfectly.

Pitmit, 2022-06-22:
[image]

fixpc82, 2022-07-04:
Is possible install on Ubuntu 18?

chbmeyer, 2023-06-28:
Hello @mrludo001 and all others,

I'm bitten by this problem too.
When you select "No", then all three devices (eth, wlan0, wlan1) are shown and asked, and when you select "Yes", the device is not found and the error stops installation.
I'm running a freshly installed "Raspbian GNU/Linux 11 (bullseye)" (Raspberry Pi reference 2023-05-03) on a "Raspberry Pi 2 Model B Rev 1.1" and git clone'd TinyCheck.

I ran the two mentioned commands and got these answers:
```
# iw wlan0 info
command failed: No such device (-19)
```
```
# ifconfig wlan0
wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        [...]
```

So it seems that iw doesn't find wlan0 while it is known by ifconfig.
There also is this message in the output of iw:
`Do NOT screenscrape this tool, we don't consider its output stable.`

So I tried to find out more and saw one known wifi device for iw and two for ifconfig:
```
# iw dev
phy#0
	Interface wlan1
		[...]
```

```
# ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        [...]

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        [...]

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        [...]

wlan1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        [...]
```

Because of this I tried to add some drivers with http://downloads.fars-robotics.net/wifi-drivers/install-wifi, but I hadn't luck with that.

Then I ended in commenting all lines 395-417 and defining IFACE_IN and IFACE_OUT manually instead:
```
IFACE_IN=wlan0
IFACE_OUT=wlan1
```

Sadly there are other problems (can not create an AP-network while running TinyCheck), but I thought, you should know about the commands mentioned above.

Christian