KasperskyLab/TinyCheck #110: Frontend service fails to start; ImportError: cannot import name 'collections_abc' from 'six.moves' (unknown location)

claudiopolis, 2022-07-11:
First, I just stumbled across this project, and want to say thanks to all the contributors. Look forward to deploying it in Thailand at the NGO that serves at risk women where I volunteer. 

I just installed this on a RasPi 4, fresh install of Raspbian Buster (2022-04-04). 

All works OK except the front-end service fails. When I run the following startup command to see why, I get the trace below. 

`sudo /usr/bin/python3 /usr/share/tinycheck/server/frontend/main.py`

```
Traceback (most recent call last):
  File "/usr/share/tinycheck/server/frontend/main.py", line 9, in <module>
    from app.blueprints.save import save_bp
  File "/usr/share/tinycheck/server/frontend/app/blueprints/save.py", line 5, in <module>
    from app.classes.save import Save
  File "/usr/share/tinycheck/server/frontend/app/classes/save.py", line 4, in <module>
    import pyudev
  File "/usr/local/lib/python3.7/dist-packages/pyudev/__init__.py", line 48, in <module>
    from pyudev.core import Context, Enumerator
  File "/usr/local/lib/python3.7/dist-packages/pyudev/core.py", line 39, in <module>
    from pyudev.device import Devices
  File "/usr/local/lib/python3.7/dist-packages/pyudev/device/__init__.py", line 26, in <module>
    from ._device import Attributes, Device, Devices, Tags
  File "/usr/local/lib/python3.7/dist-packages/pyudev/device/_device.py", line 37, in <module>
    from six.moves import collections_abc
ImportError: cannot import name 'collections_abc' from 'six.moves' (unknown location)
```

Using pip to upgrade the six package to V1.16.0 was no help. 

Would appreciate any tips, pointers etc. from anyone in the community. TIA.

claudiopolis, 2022-07-11:
I took a chance and imaged Raspbian Bullseye instead, but otherwise followed the instructions. Everything worked great. So, just the documentation needs to be updated perhaps?

mrludo001, 2022-07-15:
I can confirm this works. Use Raspbian Bullseye OS Full (32 bit). When I tried using Buster as the documentation suggests it the installation failed to find the zeek package and the installation stalls trying to update the IOC list.

clhuby, 2022-07-29:
I think this is related to the Zeek package being renamed to "zeek-lts" on Buster, this breaks the install script but it doesn't show any errors until you try to run Tinycheck. On Bullseye the install script successfully installs Zeek as it is still just called "zeek" on Bullseye.

https://github.com/KasperskyLab/TinyCheck/issues/97#issuecomment-1198890442