KasperskyLab/TinyCheck #116: IOCs detection issue ?

loromire, 2022-10-12:
Hi,
I installed TinyCheck and try it on a phone where I installed a stalkware.
Tinycheck didnt find any IOC after 20 minutes of capture.
So my question, when TinyCheck is installed, is there a list of IOCs in it ? Or do I hvae to manually put IOCs list or something like that ?
Thanks for responding guys !
Btw this soft has great potential thx Felix !

Malpaga, 2022-10-18:
Hi,
I am facing the same problem after analyzing 20 minutes of capture from a purposefully compromised phone.
An SQLite database is included with TinyCheck, you can find it in the parent directory for the app (/usr/share/TinyCheck/tinycheck.sqlite3). Apparently it already has around 4,2k IOC entries, mostly domain names.
After verification, the domain name of the server reached by the tested stalkerware was found in the sqlite ioc database.
Analysis still failed however and no problem was found within the tested device.

I will search a bit more and keep you informed if I find anything !

enricoDec, 2023-01-28:
As @Malpaga mentioned, TinyCheck comes with a default list of IOC, which can be manually extended. In the [Wiki](https://github.com/KasperskyLab/TinyCheck/wiki/TinyCheck-backend#importing-iocs-or-whitelisted-elements) you can see how to add new IOC.
I would recommend testing if TinyChecks analysis is working by manually starting it (described how-to in the Wiki [here](https://github.com/KasperskyLab/TinyCheck/wiki/TinyCheck-analysis-workflow)). In the past I had it failing and not reporting it in the frontend, but by manually starting it in the command line you can check if errors are thrown (in my case Zeek was not installed).

EvgenyAblesov, 2023-08-17:
Hello everyone in this thread!

[loromire](https://github.com/loromire), please provide more information about device you running on?

If you experiencing some troubles with 32-bit version on RPi4, please refer to https://forums.raspberrypi.com/viewtopic.php?t=351727

Long story short: RPi4 + 32-bit OS --> add "arm_64bit=0" line with no quotes to the end of your **/boot/config.txt**