KasperskyLab/TinyCheck #117: analysis.py need pango-1.0-0

0xDRRB, 2022-10-17:
Hi,

On a Raspberry Pi OS Lite installation, in headless mode, the report is not generated because `analysis.py` dies. By executing the script by hand we see that the error comes from a missing lib:

```
tinycheck:~$ sudo /usr/bin/python3 /usr/share/tinycheck/analysis/analysis.py -f /tmp/BE6E0A7A
Traceback (most recent call last):
  File "/usr/share/tinycheck/analysis/analysis.py", line 6, in <module>
    from classes.report import Report
  File "/usr/share/tinycheck/analysis/classes/report.py", line 1, in <module>
    import weasyprint
  File "/usr/local/lib/python3.9/dist-packages/weasyprint/__init__.py", line 336, in <module>
    from .css import preprocess_stylesheet  # noqa isort:skip
  File "/usr/local/lib/python3.9/dist-packages/weasyprint/css/__init__.py", line 25, in <module>
    from . import computed_values, counters, media_queries
  File "/usr/local/lib/python3.9/dist-packages/weasyprint/css/computed_values.py", line 9, in <module>
    from ..text.ffi import ffi, pango, units_to_double
  File "/usr/local/lib/python3.9/dist-packages/weasyprint/text/ffi.py", line 401, in <module>
    pango = _dlopen(
  File "/usr/local/lib/python3.9/dist-packages/weasyprint/text/ffi.py", line 385, in _dlopen
    return ffi.dlopen(names[0])  # pragma: no cover
  File "/usr/local/lib/python3.9/dist-packages/cffi/api.py", line 150, in dlopen
    lib, function_cache = _make_ffi_library(self, name, flags)
  File "/usr/local/lib/python3.9/dist-packages/cffi/api.py", line 832, in _make_ffi_library
    backendlib = _load_backend_lib(backend, libname, flags)
  File "/usr/local/lib/python3.9/dist-packages/cffi/api.py", line 827, in _load_backend_lib
    raise OSError(msg)
OSError: cannot load library 'pango-1.0-0': pango-1.0-0: cannot open shared object file: No such file or directory.  Additionally, ctypes.util.find_library() did not manage to locate a library called 'pango-1.0-0'
```
The `libpango1.0-0` package and its dependencies should be installed by `install.sh`.

loromire, 2022-10-18:
Hello,
You have to use **Raspi OS buster full version** not the lite or the extended. Only the full version works fine.
You should take raspi OS buster full from raspi imager and retry the installation.

0xDRRB, 2022-10-19:
That's not the point. Installing libpango1.0-0 fixes the problem. I just suggest including this package (and its dependencies) in the installer.