KasperskyLab/TinyCheck #122: Installation stuck

andre-santos1, 2022-11-17:
Installation stucks when it is supposed to download IoC and whitelist

carlosm2, 2023-01-07:
How much time did you wait? 
Seems that my installation is stuck too.
using a raspberry PI OS buster with a raspberry pi 3+

carlosm2, 2023-01-26:
update: I have instaled several times, and at the end it get stuck downloading IOCS
but
The IOCS were complete, check them by manually comparing the IOCS

I successfully install with raspberry pi 3 B+, Debian bullseye (full version)
and before running the installation script I install zeek of the suse repo

https://software.opensuse.org//download.html?project=security%3Azeek&package=zeek

(as mentioned here: https://github.com/KasperskyLab/TinyCheck/issues/97#issuecomment-1198890442)

Now is working fine