KasperskyLab/TinyCheck #123: Analyze Error

colder1989, 2022-11-26:
When i try to analyze the pcap i have this error in loop:

> 192.168.1.2 - - [26/Nov/2022 20:40:02] "GET /api/analysis/report/08378CB8 HTTP/1.1" 200 -
192.168.1.2 - - [26/Nov/2022 20:40:03] "GET /api/analysis/report/08378CB8 HTTP/1.1" 200 -
192.168.1.2 - - [26/Nov/2022 20:40:03] "GET /api/analysis/report/08378CB8 HTTP/1.1" 200 -
192.168.1.2 - - [26/Nov/2022 20:40:04] "GET /api/analysis/report/08378CB8 HTTP/1.1" 200 -
192.168.1.2 - - [26/Nov/2022 20:40:04] "GET /api/analysis/report/08378CB8 HTTP/1.1" 200 -
192.168.1.2 - - [26/Nov/2022 20:40:05] "GET /api/analysis/report/08378CB8 HTTP/1.1" 200 -
192.168.1.2 - - [26/Nov/2022 20:40:05] "GET /api/analysis/report/08378CB8 HTTP/1.1" 200 -
192.168.1.2 - - [26/Nov/2022 20:40:06] "GET /api/analysis/report/08378CB8 HTTP/1.1" 200 -
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/share/tinycheck/analysis/analysis.py", line 28, in zeekengine
    zeek.start_zeek()
  File "/usr/share/tinycheck/analysis/classes/zeekengine.py", line 456, in start_zeek
    self.files_check(self.working_dir + "/assets/")
  File "/usr/share/tinycheck/analysis/classes/zeekengine.py", line 272, in files_check
    "ip_src": record["tx_hosts"],
KeyError: 'tx_hosts'
Traceback (most recent call last):
  File "/usr/share/tinycheck/analysis/analysis.py", line 89, in <module>
    analyze(sys.argv[2], True)
  File "/usr/share/tinycheck/analysis/analysis.py", line 61, in analyze
    for alert in (alerts["zeek"] + alerts["suricata"]):
  File "<string>", line 2, in __getitem__
  File "/usr/lib/python3.9/multiprocessing/managers.py", line 824, in _callmethod
    raise convert_to_error(kind, result)
KeyError: 'zeek'
[2022-11-26 20:40:09,545] ERROR in app: Exception on /api/analysis/report/08378CB8 [GET]
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python3/dist-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/share/tinycheck/server/frontend/app/blueprints/analysis.py", line 29, in api_report_analysis
    return jsonify(Analysis(token).get_report())
  File "/usr/share/tinycheck/server/frontend/app/classes/analysis.py", line 59, in get_report
    alerts = json.load(f)
  File "/usr/lib/python3.9/json/__init__.py", line 293, in load
    return loads(fp.read(),
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
192.168.1.2 - - [26/Nov/2022 20:40:09] "GET /api/analysis/report/08378CB8 HTTP/1.1" 500 -
[2022-11-26 20:40:09,568] ERROR in app: Exception on /api/analysis/report/08378CB8 [GET]
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python3/dist-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/share/tinycheck/server/frontend/app/blueprints/analysis.py", line 29, in api_report_analysis
    return jsonify(Analysis(token).get_report())
  File "/usr/share/tinycheck/server/frontend/app/classes/analysis.py", line 59, in get_report
    alerts = json.load(f)
  File "/usr/lib/python3.9/json/__init__.py", line 293, in load
    return loads(fp.read(),
  File "/usr/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

colder1989, 2022-11-26:
i think the problem might be zeek

vollkorn1982, 2022-12-17:
Running into the same problem. It seems like the log file format of zeek has changed. The offending python code is checking trying to find `tx_hosts` in `/tmp/<id>/assets/files.log`, but that string isn't in there.

vollkorn1982, 2022-12-17:
I changed lines 272 and 274 in `/usr/share/tinycheck/analysis/classes/zeekengine.py` like this:

```
          c = {"ip_dst": record["id.resp.h"],
                 ...
                 "port_dst": record["id.resp_p"],
```
Then I was able to run the analysis manually by calling `sudo python3 /usr/share/tinycheck/analysis/analysis.py /tmp/<id>/` and found results in alerts.json.

I'll create a PR tomorrow.

commit 7696f840 (2023-01-11): Merge pull request #124 from vollkorn1982/fix_failing_zeek_analysis

zeek 5.1.1-0 has a new logfile format, fixes #123
commit fae904ec (2023-01-11): zeek 5.1.1-0 has a new logfile format, fixes #123

The columns tx_host and rx_host have been renamed. This commit reflects the
name change and repairs the analysis process.