KasperskyLab/TinyCheck #125: Installation instructions in wiki are outdated

vollkorn1982, 2022-12-18:
The Wiki states on https://github.com/KasperskyLab/TinyCheck/wiki/TinyCheck-installation that the installation is only tested on Raspberry Buster. This does not work as of at least today (2022-12-18). This is because zeek is not offered for Raspbian Buster anymore, as can be seen by checking the download path http://download.opensuse.org/repositories/security:/zeek/Raspbian_10/ in a browser:

```
Objekt nicht mehr verfügbar!

Der angeforderte URL existiert auf dem Server nicht mehr und wurde dauerhaft entfernt. Eine Weiterleitungsadresse ist nicht verfügbar. Falls Sie einem Link von einer anderen Seite gefolgt sind, informieren Sie bitte den Autor dieser Seite hierüber.

Sofern Sie dies für eine Fehlfunktion des Servers halten, informieren Sie bitte den [Webmaster](mailto:webmaster@opensuse.org) hierüber.
Error 404
[download.opensuse.org](http://download.opensuse.org/)
Apache
```
I would update the occurrences of "Buster" to "Bullseye". I just tested the installation and operation of TinyCheck on Raspbian 11 Bullseye and had no problems, aside my little fix in PR #124.

alexsimon1, 2023-01-03:
Hi @vollkorn1982 did you had the same issue with cryptography ? I'm using bullseye 32 bit lite os.

`Building wheels for collected packages: cryptography
  Building wheel for cryptography (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /tmp/tmpkve59qoj_in_process.py build_wheel /tmp/tmp1_7v97ph
       cwd: /tmp/pip-install-_yihi6tw/cryptography_2f587e5047d34e779357da34b1d54f7b
  Complete output (191 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-cpython-39
  creating build/lib.linux-armv7l-cpython-39/cryptography
  copying src/cryptography/fernet.py -> build/lib.linux-armv7l-cpython-39/cryptography
  copying src/cryptography/exceptions.py -> build/lib.linux-armv7l-cpython-39/cryptography
  copying src/cryptography/__about__.py -> build/lib.linux-armv7l-cpython-39/cryptography
  copying src/cryptography/utils.py -> build/lib.linux-armv7l-cpython-39/cryptography
  copying src/cryptography/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography
  creating build/lib.linux-armv7l-cpython-39/cryptography/hazmat
  copying src/cryptography/hazmat/_oid.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat
  copying src/cryptography/hazmat/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat
  creating build/lib.linux-armv7l-cpython-39/cryptography/x509
  copying src/cryptography/x509/ocsp.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
  copying src/cryptography/x509/name.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
  copying src/cryptography/x509/oid.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
  copying src/cryptography/x509/general_name.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
  copying src/cryptography/x509/base.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
  copying src/cryptography/x509/extensions.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
  copying src/cryptography/x509/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
  copying src/cryptography/x509/certificate_transparency.py -> build/lib.linux-armv7l-cpython-39/cryptography/x509
  creating build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/keywrap.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/padding.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/hashes.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/_asymmetric.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/constant_time.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/poly1305.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/_serialization.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/_cipheralgorithm.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/cmac.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  copying src/cryptography/hazmat/primitives/hmac.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives
  creating build/lib.linux-armv7l-cpython-39/cryptography/hazmat/backends
  copying src/cryptography/hazmat/backends/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/backends
  creating build/lib.linux-armv7l-cpython-39/cryptography/hazmat/bindings
  copying src/cryptography/hazmat/bindings/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/bindings
  creating build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives/asymmetric
  copying src/cryptography/hazmat/primitives/asymmetric/dsa.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives/asymmetric
  copying src/cryptography/hazmat/primitives/asymmetric/ed448.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives/asymmetric
  copying src/cryptography/hazmat/primitives/asymmetric/rsa.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives/asymmetric
  copying src/cryptography/hazmat/primitives/asymmetric/padding.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives/asymmetric
  copying src/cryptography/hazmat/primitives/asymmetric/utils.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives/asymmetric
  copying src/cryptography/hazmat/primitives/asymmetric/ed25519.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives/asymmetric
  copying src/cryptography/hazmat/primitives/asymmetric/types.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives/asymmetric
  copying src/cryptography/hazmat/primitives/asymmetric/dh.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives/asymmetric
  copying src/cryptography/hazmat/primitives/asymmetric/__init__.py -> build/lib.linux-armv7l-cpython-39/cryptography/hazmat/primitives/asymmetric
[... cut ...]
  If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.

  This package requires Rust >=1.48.0.
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
`

vollkorn1982, 2023-01-04:
No, I did not have this problem. I recommend using the "normal" Raspbian flavor and not the lite version. It comes with quite a few dependencies pre-installed, that seem not to be covered by the TinyCheck install script.

chebatory, 2023-01-11:
I removed reference to buster from wiki.