KasperskyLab/TinyCheck #127: Unable to access the network

pasqualepalmaccio, 2023-01-02:
I have a Raspberry Pi 4 with Tinycheck installed. The program starts correctly and creates the SSID-hash network, for example "network-abcd". The created network is not visible on any of my mobile devices and if I try to connect using the QR code, it says "unable to access the network." I have tried many times even by manually entering the connection parameters.
I have tried on my iPhone and my Android phone with same result 

I'm using eth0 for internet connection and wlan0 for AP mode.

EvgenyAblesov, 2023-10-03:
Hey @pasqualepalmaccio, can you please reinstall TinyCheck and provide some logs?

SaTosh1, 2023-12-08:
Did you ever get this fixed? having the same issue.