KasperskyLab/TinyCheck #128: Issue with the convert_unicode argument on SQLAlchemy

Bunoud, 2023-01-27:
Hello, 
I made a fresh install on rasp pi 3B of the Bulleye 32bit full OS. 
The installation was fine with no error code. When rebooting for first tests, it seems that the front end server did not start. 
I tried to restart the frontend process and saw an issue in __init__.py in line 10 
It seems that the "convert_uncicode=True" is no longer usable. 
I've tried to figure out what was the origine of this issue, and tried to downgrade the version of SQLAlchemy to 1.4.46 with no success. 
Do you had some hints for this issue? 
Thanks

Bunoud, 2023-01-30:
After few investigations and some try on debugging its seems that the whole process of mapping and charging the DB is no longer supported by SQLAlchemy. Am i the only one to get this issue ? 
Thanks !

tdmdu31, 2023-02-07:
Hello
i have the same problem after trying some other linux versions

Bunoud, 2023-02-09:
I have tried to downgrade to differents versions, but it seems that no one is working .. 
Regards,

AlexanderKgr, 2023-02-10:
the same issue I have also faced. 
anyone found a workaround?

AlexanderKgr, 2023-02-10:
found a solution.
edit the file /assets/requirements.txt

replace
sqlalchemy
with 
sqlalchemy==1.4.46

uninstall tinyckeck reboot install it again and will work

Bunoud, 2023-02-10:
It's worked for me too. Thanks

ldina, 2023-02-27:
I solved the services not starting by sudo pip install SQLAlchemy==1.4.46 without reinstalling tinicheck

EvgenyAblesov, 2023-08-17:
Hello everyone!

Seems like problem was solved prior to be fixed.
https://github.com/KasperskyLab/TinyCheck/blob/main/assets/requirements.txt

closed 2023-08-17 by EvgenyAblesov