KasperskyLab/TinyCheck #131: 127.0.0.1 refused to connect

howky78, 2023-02-26:
Hi,
Total noob at this coding. 
I managed to turn the Pi 4 into an AP and everything still worked.
Then installed TinyCheck and installed fine…
Once rebooted I get “site can’t be reached..127.0.0.1 refused to connect’
Is there something I’ve missed?
Any help appreciated and if put in easy to follow guide would be appreciated.

ldina, 2023-02-27:
I'm not sure if you are in my same situation but I solved the services not starting by sudo pip install SQLAlchemy==1.4.46

howky78, 2023-02-27:
> I'm not sure if you are in my same situation but I solved the services not starting by sudo pip install SQLAlchemy==1.4.46

Is that after the install? 
im trying to work out if I have to do anything to the Wi-Fi settings prior to the tinycheck install. 
The wiki said it was simple to install but it’s far from it I’m my case lol

ldina, 2023-02-27:
> > I'm not sure if you are in my same situation but I solved the services not starting by sudo pip install SQLAlchemy==1.4.46
 
> Is that after the install? 
yes

howky78, 2023-02-28:
> > > I'm not sure if you are in my same situation but I solved the services not starting by sudo pip install SQLAlchemy==1.4.46
> 
> > Is that after the install?
> > yes

Thank you.
Did you have to do anything to the Wi-Fi setup prior to the tinycheck install? I have the Pi connected by Ethernet

ldina, 2023-02-28:
no, my wifi setup worked with no issues. Just plugged in. I'm using an Alfa wifi device

howky78, 2023-02-28:
Ok thanks, I’ve tried a Wi-Fi adapter but didn’t work. And if I leave the Wi-Fi setting as is, when it asks for Eth0 and wlan0 to be the front and back end it fails. I’ll try again later and find the right wording

ldina, 2023-02-28:
A limited number of wifi adapter is supported. Be sure to have the adapter working before trying to install tinycheck

howky78, 2023-02-28:
Thanks, one ordered from the pihut. Will wait for that then try again. 
thanks

howky78, 2023-03-01:
> install

Thanks for your help.
All sorted and working fine.

ppuschmann, 2023-05-06:
Installing SQLAlchemy exactly did help.

```
sudo pip install SQLAlchemy==1.4.46
```
Then Reboot, everything is fine :-)

EvgenyAblesov, 2023-06-29:
Thanks for your answers and help.
We're now using 'version pinning' in installation process for python modules.

closed 2023-06-29 by EvgenyAblesov