KasperskyLab/TinyCheck #132: Install: "You must select two interfaces, exiting"

mousesonmarses, 2023-03-29:
Hi,
im getting: "You must select two interfaces, exiting", whil trying to install.
The raspi has a fixed ip on eth0 and dhcp on wlan0 and wlan1.
[image]

omer-re, 2023-04-11:
happens to me too, any solution yet?

ldina, 2023-04-11:
I noted that during the installation the installer get confused. The best way is to have eth0 as a bridge to the internet and one unconnected wlan as AP. When the installation is completed you can change it to whatever mix you like. I was successful with a raspberry eth0+wlan0 (external usb) or a PC wlan0+wlan1 (external usb) and eth0 down

icedan, 2023-11-21:
apt install rfkill
rfkill block wifi

and now is working

EvgenyAblesov, 2023-12-08:
`apt install rfkill`
`rfkill block wifi`

We will try to improve installation process of TinyCheck to avoid such situations in the future. Thanks for workaround

closed 2023-12-08 by EvgenyAblesov