KasperskyLab/TinyCheck #138: How to connect and capture multiple devices

dawidol, 2023-06-24:
Hi 

I want to connect several devices to the tinycheck, is it possible? the normal behaviour look like after one device is connected the capture start. 

It is just requiered to add another device to the network and its traffic will be also captured or do I need to do it in any other way.

Thanks a lot

EvgenyAblesov, 2023-06-26:
The design of the TinyCheck application does not allow the use of multiple parallel connections. 

This is primarily due to the performance provided by Rasperry Pi. To understand the limitations of the platform just try using one device for about 10 or 15 minutes. The size of the capture file for futher analysis from a single device can be significant so adding support for multiple devices would be a technical problem.

closed 2023-06-26 by EvgenyAblesov

dawidol, 2023-06-26:
Thank you for your reply. Yes I understand the idea, you think it is valid using a raspberry pi 4b or even a laptop with enough resources?  I am using a RPi 4b with 8 Gb of RAM, and been monitoring the performance with tiny check and is very low.