KasperskyLab/TinyCheck #146: Issues while installing TinyCheck on a fresh Ubuntu 20.04. live usb

reetuh797, 2023-12-10:
Hello, 

I have a couple of issues. Please help. If you can answer in detail, that would be  helpful, as I'am not experienced with terminal or IT in general.

I'm using: 
a mobile hotspot,
a laptop PC with an inbuilt router, that doesn't support AP mode, 
and a  TPLink N 300 adapter, which does support Soft-AP as the main source of internet.

Context: 
After finally managing to install TinyCheck, I couldn't connect nor to the frontend nor to the backend. I tried to reinstall - I uninstalled TinyCheck, found [this answer](https://github.com/KasperskyLab/TinyCheck/issues/114) and installed by myself nodejs, and then tried reinstalling TinyCheck hoping to find instructions and follow the aforementioned advice, but now I see that a the moment the installation is building in the right directories ... I hadn't checked that before, so installing nodejs beforehand wasn't necessary as I understand now.

Anyways, I now get the following two issues A and B:

A. Deprecation warnings, before the installation proceeds. I don't remember, if I had them pop up the last time, maybe not, because nodejs was being installed properly as a part of the TinyCheck installation process.  

The Warnings: 
1)
"DEPRECATION WARNING 
Node.js 14.x is no longer actively supported!You will not receive security or critical stability updates for this version.You should migrate to a supported version of Node.js as soon as possible.Use the installation script that corresponds to the version of Node.js youwish to install. e.g.* https://deb.nodesource.com/setup_16.x — Node.js 16 "Gallium"* https://deb.nodesource.com/setup_18.x — Node.js 18 LTS "Hydrogen" (recommended)* https://deb.nodesource.com/setup_19.x — Node.js 19 "Nineteen"* https://deb.nodesource.com/setup_20.x — Node.js 20 "Iron" (current)Please see https://github.com/nodejs/Release for details about whichversion may be appropriate for you.The NodeSource Node.js distributions repository containsinformation both about supported versions of Node.js and supported Linuxdistributions. To learn more about usage, see the repository:https://github.com/nodesource/distributions"

2) "SCRIPT DEPRECATION WARNING This script, located at https://deb.nodesource.com/setup_X, used toinstall Node.js is deprecated now and will eventually be made inactive.Please visit the NodeSource distributions Github and follow theinstructions to migrate your repo.https://github.com/nodesource/distributionsThe NodeSource Node.js Linux distributions GitHub repository containsinformation about which versions of Node.js and which Linux distributionsare supported and how to install it.https://github.com/nodesource/distributionsSCRIPT DEPRECATION WARNING"

B. nmp audit warnings and fix'es , which I remember seeing the last time:

The fix'es:

"found 198 vulnerabilities (2 low, 89 moderate, 83 high, 24 critical)
  run `npm audit fix` to fix them, or `npm audit` for details
npm WARN deprecated @babel/plugin-proposal-class-properties@7.18.6: This proposal has been merged to the ECMAScript standard and thus this plugin is no longer maintained. Please use @babel/plugin-transform-class-properties instead.
npm WARN deprecated chokidar@2.1.8: Chokidar 2 does not receive security updates since 2019. Upgrade to chokidar 3 with 15x fewer dependencies
npm WARN deprecated fsevents@1.2.13: The v1 package contains DANGEROUS / INSECURE binaries. Upgrade to safe fsevents v2
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.2.7 (node_modules/webpack-dev-server/node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN @babel/plugin-bugfix-v8-spread-parameters-in-optional-chaining@7.23.3 requires a peer of @babel/core@^7.13.0 but none is installed. You must install peer dependencies yourself
npm WARN @babel/plugin-transform-class-static-block@7.23.4 requires a peer of @babel/core@^7.12.0 but none is installed. You must install peer dependencies yourself.
npm WARN @kaspersky/tinycheck-backend@0.1.0 No repository field.
npm WARN @kaspersky/tinycheck-backend@0.1.0 No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules/watchpack-chokidar2/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})"

  [....] "╭────────────────────────────────────────────────────────────────╮
   │                                                                │
   │      New major version of npm available! 6.14.18 → 10.2.5      │
   │   Changelog: https://github.com/npm/cli/releases/tag/v10.2.5   │
   │               Run npm install -g npm to update!                │
   │                                                                │
   ╰────────────────────────────────────────────────────────────────╯

> @kaspersky/tinycheck-backend@0.1.0 build /usr/share/tinycheck/app/backend
> vue-cli-service build
"
[...]

the same npm warnings are displayed while building the frontend.
Here, I also think that the outdated npm is due to me installing nodejs by myself, so that's on me, but I suspect that the connection problem might e related to the issues detected during npm audit. Might that be true? 

(Sorry if I'm writing non-sense).

Current state: 
Otherwise all of the boxes were checked green during the installation. And the system reboots. 

I will update.

reetuh797, 2023-12-15:
I managed to get up until connecting to the ephemeral network. When it gets created my internet connection disappears and my WIFI icon turns green. When I scan the QR code after some time my computer gets back online, the connected phone stays offline. One time the program even started the analysis, but then the phone went offline.

I think the problem is that the program is reading my inbuilt router s the wireless interface and my usb adapter as the internet interface and it may need to be the other way around, because the inbuilt router doesn't support AP, while as it turns out it does support master mode.

@EvgenyAblesov, you have been active here recently, so if I may ask, perhaps you know how to change the devices? Or what the issue might be here? (Backend doesn't allow me to choose anything else for the WiFi interface than my inbuilt router).
 Sorry to be bothersome. I really need help.

EvgenyAblesov, 2023-12-15:
> Caution: This is an external email. Be cautious while opening links or attachments.

Hi!

I have read all your messages and that's what I have to say about it.

Unfortunately, you are not using TinyCheck in the way that it was designed and designed for.
TinyCheck never had Ubuntu compatibility in general and designed to run only on Raspbian 32bit.
So, is a huge job that you have already done!
And I really can't help you just because I don't have the same equipment as you do.
Also my network settings will be significantly different from yours.

However, here's what I'll do in this weekend. I will install Ubuntu on a virtual machine and connect Wi-Fi router to it. Please notice, that my model of Wi-Fi router can be significally differs from yours. But this one is the only one that I have available. I will also set up the network as it is customary in my region and fits to my provider requrements.

Then I will try to solve all the compatibility problems and make a list of such problem-solutions for you. Maybe, somehow some points from this list will suit you. If I find any common solutions between Ubuntu and Raspbian, I will make a new pull request on GitHub and mention it in my next reply to you.
I plan to do this this weekend, is it ok to get an answer next week for you?
Will you be satisfied with the lack of a solution in the end?

Спасибо!

Office: +7 495 797 87 00 x1659 | Mobile: +7 926 245 11 20 | ***@***.******@***.***>
39A/3, Leningradskoe Shosse, Moscow, 125212, Russia | www.kaspersky.com<http://www.kaspersky.com>
blog.kaspersky.com<http://blog.kaspersky.com/> | business.kaspersky.com<http://business.kaspersky.com/> | securelist.com<http://securelist.com/> | threatpost.com<http://www.threatpost.com/>

reetuh797, 2023-12-15:
[@EvgenyAblesov](https://github.com/EvgenyAblesov) 
Huge thanks for answering my question! I will be happy for any help you have time for. 

It is only thanks to the answers here and in the ubuntu community, I only had to struggle through pasting the code and hoping for the best.
I understand that the program wasn't made for these devices. But there isn't another option available. Except maybe the MVT, which I haven't tried yet.

Best regards, 
ReeTuh